zephyrproject-rtos/zephyr #10636: quark_se is missing SPI_DRV_NAME symbol

finikorg, 2018-10-16:
After commit 55caa7b7434e769c5da2680a5161da8ec7caa260 SPI_1_NAME is empty for a board quark_se_c1000_devboard

finikorg, 2018-10-16:
@tbursztyka

MaureenHelm, 2018-10-16:
The Kconfig symbol `SPI_1_NAME` is intentionally undefined for this board after this commit.  The preprocessor macro `CONFIG_SPI_1_NAME` is still defined, but now it's generated from dts instead of Kconfig.

Excerpts from `tests/drivers/spi/spi_loopback/build/zephyr/include/generated/generated_dts_board.h`:
```c
/* spi@b0001400 */
#define SNPS_DESIGNWARE_SPI_B0001400_BASE_ADDRESS       0xb0001400
#define SNPS_DESIGNWARE_SPI_B0001400_IRQ_0              3
#define SNPS_DESIGNWARE_SPI_B0001400_IRQ_0_PRIORITY     2
#define SNPS_DESIGNWARE_SPI_B0001400_IRQ_0_SENSE        32768
#define SNPS_DESIGNWARE_SPI_B0001400_LABEL              "SPI_1"                                
#define SNPS_DESIGNWARE_SPI_B0001400_SIZE               1024

#define CONFIG_SPI_1_BASE_ADDRESS       SNPS_DESIGNWARE_SPI_B0001400_BASE_ADDRESS
#define CONFIG_SPI_1_NAME               SNPS_DESIGNWARE_SPI_B0001400_LABEL                     
#define CONFIG_SPI_1_IRQ                SNPS_DESIGNWARE_SPI_B0001400_IRQ_0
#define CONFIG_SPI_1_IRQ_PRI            SNPS_DESIGNWARE_SPI_B0001400_IRQ_0_PRIORITY
```

finikorg, 2018-10-17:
@MaureenHelm so should the functionality be the same as before? At the moment SPI_NAME is undefined (whatever it is called now)

MaureenHelm, 2018-10-17:
You can't change `SPI_1_NAME` value in menuconfig anymore, but you can use `CONFIG_SPI_1_NAME` just as you did before:

https://github.com/zephyrproject-rtos/zephyr/blob/92080a0670a8b1b7a6808688057665cc58b9804f/drivers/spi/spi_dw.c#L594-L597

Is there a particular sample or test application that's broken?

finikorg, 2018-10-18:
`samples/net/wpanusb` is not working anymore

MaureenHelm, 2018-10-18:
Ok, I see the problem now. We need to add dts bindings for the cc2520 like we did for the mcr20a.

galak, 2018-10-29:
There are a few cleanups like this we need to make, will open another issue to catch them all.

MaureenHelm, 2019-01-28:
Looks like this was fixed in #11428. @finikorg 	can you confirm?

finikorg, 2019-01-30:
> Looks like this was fixed in #11428. @finikorg can you confirm?

@MaureenHelm  Works now, thanks!

MaureenHelm, 2019-01-30:
Fixed by #11428

closed 2019-01-30 by MaureenHelm