zephyrproject-rtos/zephyr #10639: make run for native_posix has dependency issues

nashif, 2018-10-16:
call 'make run' for native_posix and the build will fail, we do not have dependency of the run taget on the .exe binary.

```
Scanning dependencies of target run
make[3]: Leaving directory '/home/nashif/Work/github/zephyr/build'
make -f zephyr/CMakeFiles/run.dir/build.make zephyr/CMakeFiles/run.dir/build
make[3]: Entering directory '/home/nashif/Work/github/zephyr/build'
/home/nashif/Work/github/zephyr/build/zephyr/zephyr.exe
make[3]: /home/nashif/Work/github/zephyr/build/zephyr/zephyr.exe: Command not found
make[3]: *** [zephyr/CMakeFiles/run.dir/build.make:60: zephyr/CMakeFiles/run] Error 127
make[3]: Leaving directory '/home/nashif/Work/github/zephyr/build'
make[2]: *** [CMakeFiles/Makefile2:271: zephyr/CMakeFiles/run.dir/all] Error 2
make[2]: Leaving directory '/home/nashif/Work/github/zephyr/build'
make[1]: *** [CMakeFiles/Makefile2:278: zephyr/CMakeFiles/run.dir/rule] Error 2
make[1]: Leaving directory '/home/nashif/Work/github/zephyr/build'
make: *** [Makefile:173: run] Error 2
make: Leaving directory '/home/nashif/Work/github/zephyr/build'

```

nashif, 2018-10-16:
actually that does not solve the problem because we are renaming...

```
cd /home/nashif/Work/github/zephyr/build/zephyr && /usr/bin/cmake -E rename zephyr.elf zephyr.exe
make[3]: Leaving directory '/home/nashif/Work/github/zephyr/build'
[100%] Built target zephyr_prebuilt
make -f zephyr/CMakeFiles/run.dir/build.make zephyr/CMakeFiles/run.dir/depend
make[3]: Entering directory '/home/nashif/Work/github/zephyr/build'
cd /home/nashif/Work/github/zephyr/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/nashif/Work/github/zephyr/samples/synchronization /home/nashif/Work/github/zephyr /home/nashif/Work/github/zephyr/build /home/nashif/Work/github/zephyr/build/zephyr /home/nashif/Work/github/zephyr/build/zephyr/CMakeFiles/run.dir/DependInfo.cmake --color=
Dependee "/home/nashif/Work/github/zephyr/build/zephyr/CMakeFiles/run.dir/DependInfo.cmake" is newer than depender "/home/nashif/Work/github/zephyr/build/zephyr/CMakeFiles/run.dir/depend.internal".
Dependee "/home/nashif/Work/github/zephyr/build/zephyr/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/nashif/Work/github/zephyr/build/zephyr/CMakeFiles/run.dir/depend.internal".
Scanning dependencies of target run
make[3]: Leaving directory '/home/nashif/Work/github/zephyr/build'
make -f zephyr/CMakeFiles/run.dir/build.make zephyr/CMakeFiles/run.dir/build
make[3]: Entering directory '/home/nashif/Work/github/zephyr/build'
make[3]: *** No rule to make target 'zephyr/zephyr.elf', needed by 'zephyr/CMakeFiles/run'.  Stop.
make[3]: Leaving directory '/home/nashif/Work/github/zephyr/build'
make[2]: *** [CMakeFiles/Makefile2:271: zephyr/CMakeFiles/run.dir/all] Error 2
make[2]: Leaving directory '/home/nashif/Work/github/zephyr/build'
make[1]: *** [CMakeFiles/Makefile2:278: zephyr/CMakeFiles/run.dir/rule] Error 2
make[1]: Leaving directory '/home/nashif/Work/github/zephyr/build'
make: *** [Makefile:173: run] Error 2
make: Leaving directory '/home/nashif/Work/github/zephyr/build'

```

pfalcon, 2018-10-19:
Similar report from me: https://github.com/zephyrproject-rtos/zephyr/pull/10640#issuecomment-430689014

pfalcon, 2018-10-19:
Then following @SebastianBoe 's suggestion in https://github.com/zephyrproject-rtos/zephyr/pull/10429#issuecomment-431260571 to try https://github.com/SebastianBoe/zephyr/commit/0c72c82fdad67bef68cc6ecaa0d9c9a170ffcae0 , that commit fixes it for me, so I suggest to submit it as PR (if not yet).

PR #10640 (2018-10-16): cmake: Fix 'run' command of native_posix
The 'run' target was missing it's dependency on the executable and
thereby behaving incorrectly.

This fixes #10639 .
commit 7f99086d (2018-10-16): cmake: Fix 'run' command of native_posix

The 'run' target was missing it's dependency on the executable and
thereby behaving incorrectly.

This fixes #10639 .
PR #10643 (2018-10-19): build: do not rename elf file, copy it
We break dependency if we rename the zephyr.elf file, we should keep it
as it is an essential dependency in the build system. Instead, copy the
file.

Fixes #10639
commit 951393f6 (2018-10-19): build: do not rename elf file, copy it

We break dependency if we rename the zephyr.elf file, we should keep it
as it is an essential dependency in the build system. Instead, copy the
file.

Fixes #10639
closed 2018-10-19 by nashif