zephyrproject-rtos/zephyr #10644: logging: Where's my function name??

pfalcon, 2018-10-16:
With new logger, I get:

~~~
[00:00:19.606,640] <err> net_tcp: context->tcp == NULL
[00:00:19.612,784] <err> net_tcp: context->tcp == NULL
[00:00:19.615,296] <err> net_tcp: context->tcp == NULL
[00:00:19.618,140] <err> net_tcp: context->tcp == NULL
~~~

What's this? How to get use of this? I need the function name where it happens, like was there with the old logger. (It was there, right, or am I hallucinating? All in all, with the old logger, it was clear (ahem) what happens where, and the new logger really leads to that question - what's this?)

jukkar, 2018-10-16:
NET_DBG() will add function name. There was no function name for error log prints with old logging.

jhedberg, 2018-10-18:
We need this also for Bluetooth (see the duplicate issue I opened by mistake). Currently #10264 is mapping the Bluetooth logging macros as follows:
```
#define BT_DBG(fmt, ...) LOG_DBG("%s: " fmt, __func__, ##__VA_ARGS__)
#define BT_ERR(fmt, ...) LOG_ERR("%s: " fmt, __func__, ##__VA_ARGS__)
#define BT_WARN(fmt, ...) LOG_WRN("%s: " fmt, __func__, ##__VA_ARGS__)
#define BT_INFO(fmt, ...) LOG_INF(fmt, ##__VA_ARGS__)
```
I'd like to completely remove the Bluetooth-specific macros and replace them with corresponding LOG_* ones however as long as the function name support is not there this would change the behavior.

PR #10708 (2018-10-24): logging: Add optional function name prefix
Extended logger to support optional log message prepending with function name.

Prepending with function name is optional (enabled by default). I've used following formatting:
`[00:00:24.229,644] <err> app_test.foo(): err message`

Fixes #10644
Fixes #10700
closed 2018-10-24 by nashif