zephyrproject-rtos/zephyr #10649: bug when build project which new c lib

pengguiyong, 2018-10-17:
r: conflicting types for ‘remove’
 static void remove(struct _timeout *t)
             ^~~~~~
In file included from /home/pgy/learn/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal_def.h:48:0,
                 from /home/pgy/learn/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal_rcc.h:45,
                 from /home/pgy/learn/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal_conf.h:259,
                 from /home/pgy/learn/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal.h:46,
                 from /home/pgy/learn/zephyr/ext/hal/st/stm32cube/stm32f4xx/soc/stm32f4xx.h:250,
                 from /home/pgy/learn/zephyr/soc/arm/st_stm32/stm32f4/soc.h:27,
                 from /home/pgy/learn/zephyr/include/arch/arm/cortex_m/cmsis.h:21,
                 from /home/pgy/learn/zephyr/arch/arm/include/cortex_m/stack.h:27,
                 from /home/pgy/learn/zephyr/arch/arm/include/kernel_arch_data.h:52,
                 from /home/pgy/learn/zephyr/kernel/include/kernel_structs.h:65,
                 from /home/pgy/learn/zephyr/kernel/include/ksched.h:10,
                 from /home/pgy/learn/zephyr/kernel/timeout.c:10:
/opt/zephyr-sdk/sysroots/armv5-zephyr-eabi/usr/include/stdio.h:227:5: note: previous declaration of ‘remove’ was here
 int _EXFUN(remove, (const char *));

pengguiyong, 2018-10-17:
redefine this function name will be ok   
remove_tt(struct _timeout *t)

nashif, 2018-10-18:
this is already fixed.

closed 2018-10-18 by nashif