zephyrproject-rtos/zephyr #10652: qustion: How to ping host target from Zephyr?

qianfan-Zhao, 2018-10-17:
Hi, I want ping a target device from Zephyr, and check if the target alive. But the API net_icmpv4_send_echo_request is declared in subsys/net/ip, not exported for us. Is there has a easy way to do that?

qianfan-Zhao, 2018-10-17:
My personal idea is based on PR #10665 and PR #10660 .

```
static bool network_check(int arg)
{
	char cmd[64] = "net ping " CONFIG_NET_CONFIG_PEER_IPV4_ADDR;

	return !shell_execute_cmd(NULL, cmd);
}
```

This maybe not the best way, but it just works.
If we can export some ICMP APIs to public, or implement SOCK_RAW?

jukkar, 2018-10-19:
Is there need to do anything here? I thought after IRC conversation that there is some progress when working with `shell_execute_cmd()`.

closed 2018-11-28 by qianfan-Zhao