zephyrproject-rtos/zephyr #10657: tests/net/ieee802154/crypto does not work as expected

nniranjhana, 2018-10-17:
**Arch:** x86
**Platform:** quark_se_c1000

It looks like the test should ideally bind with the crypto device, use the key, auth data provided by the test to verify authentication, encryption and display the results:

> Authentication only test: PASSED
> Encryption only test: PASSED
> Authentication and encryption test: PASSED

But, when test application is flashed, I only get a uart console asking for input, and then after a minute or two I get a _CPU exception_.

**Error log on console:**
```
uart:~$

***** CPU exception 6

Current thread ID = 0xa8007f60
eax: 0x1578fd4c, ebx: 0x00000000, ecx: 0xa800bd28, edx: 0xa8007444
esi: 0xa800bd38, edi: 0xa8007420, ebp: 0xa800bd54, esp: 0xa800bd18
eflags: 0x00010202 cs: 0x0008
eip: 0x40000311
Fatal fault in essential thread! Spinning...
```

galak, 2019-03-27:
@tbursztyka have you looked at this?

tbursztyka, 2019-04-01:
See #8339

tbursztyka, 2019-04-05:
@nniranjhana  can you retry now #8339 is in master? that should fix the issue.

tbursztyka, 2019-04-17:
Closing this as #14103 fixed that specific issue.

There is still an issue for the test to pass, but that's in issue #15449

closed 2019-04-17 by tbursztyka