zephyrproject-rtos/zephyr #10664: FOTA: nRF52: integration of samples/bluetooth/mesh & smp_srv

vikrant8051, 2018-10-17:
Need guidance in integration of samples/bluetooth/mesh & samples/subsys/mgmt/mcumgr/smp_svr:

Goal is to :
1) flash #mcuboot on nRF52_PDK
2) then flash signed original.hex
3) provision it using #nRFMesh
4) firmware upgrade over Bluetooth using #nRFConnect
    (by uploading signed modified.bin)
5) NODE should remain in provisioned state.

jhedberg, 2018-10-18:
How far have you gotten in these steps? At what point do you get stuck?

jhedberg, 2018-10-18:
For the record, as long as you don't change the composition of the node and don't overwrite the data in the storage partition everything should work as you describe in the steps. If you change the composition you will need to reprovision no matter what since the Mesh specification does not provide a way for nodes to change their composition after provisioning.

vikrant8051, 2018-10-18:
@jhedberg
[main.txt](https://github.com/zephyrproject-rtos/zephyr/files/2493654/main.txt)
[prj.txt](https://github.com/zephyrproject-rtos/zephyr/files/2493657/prj.txt)

To integrate smp_svr, I modified main.c & prj.conf as per attached file.
These are part of #onoff_level_lighting_vnd_app.

After these changes, I created original_signed.hex & modified_signed.bin from
same App (composition of the NODE is exactly same in these two versions..... modified_signed.bin only
contains one extra prink statement at the last of main() to differentiate it from original)

Till step 3 everything is working fine, but if I reset the board just after it then NODE get
unprovisioned. After DFU whether NODE remains in provision state or not is not concern
for me.As of now, I wanna know that why NODE gets unprovisioned after reset with original firmware ?

This means I've not properly integrate smp_svr. Could you please correct me.?

I've also observed that shell get activated on serial terminal & board takes some time to reboot.
So how to remove unnecessary things from curent integration & optimise it.

vikrant8051, 2018-11-01:
@jhedberg @carlescufi 
Today I modified #onoff_level_lighting_vnd_app's files as per attachments.
Now able to follow all 5 steps which are mentioned in introduction of this issue. 

Node remains in provisioned state after DFU.

But is it optimal integration ?
Have I done everything correctly ?

Waiting for reply since then I could create new PR to introduced #DFU_OTA 
feature in #onoff_level_lighting_vnd_app.

[prj.conf.txt](https://github.com/zephyrproject-rtos/zephyr/files/2537774/prj.conf.txt)
[main.txt](https://github.com/zephyrproject-rtos/zephyr/files/2537776/main.txt)

vikrant8051, 2018-11-05:
Hi,
waiting for some feedback.

carlescufi, 2019-08-15:
@vikrant8051 can you please send a Pull Request instead so we can discuss it there? Discussing code in an issue is quite cumbersome. If you instead have written questions please reopen this.

closed 2019-08-15 by carlescufi