zephyrproject-rtos/zephyr #10669: kernel: "New timer subsystem" uses oversimplified common names, leading to debugging confusion and compilation clashes

pfalcon, 2018-10-17:
From https://github.com/zephyrproject-rtos/zephyr/pull/10303#issuecomment-430638826 : 

~~~
/home/buildslave/workspace/lite-aeolus-micropython/zephyr/kernel/timeout.c:47:13: error: conflicting types for 'remove'
12:22:03  static void remove(struct _timeout *t)
12:22:03              ^~~~~~
12:22:03 In file included from /home/buildslave/workspace/lite-aeolus-micropython/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal_def.h:48:0,
12:22:03                  from /home/buildslave/workspace/lite-aeolus-micropython/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal_rcc.h:45,
12:22:03                  from /home/buildslave/workspace/lite-aeolus-micropython/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal_conf.h:259,
12:22:03                  from /home/buildslave/workspace/lite-aeolus-micropython/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/include/stm32f4xx_hal.h:46,
12:22:03                  from /home/buildslave/workspace/lite-aeolus-micropython/zephyr/ext/hal/st/stm32cube/stm32f4xx/soc/stm32f4xx.h:250,
12:22:03                  from /home/buildslave/workspace/lite-aeolus-micropython/zephyr/soc/arm/st_stm32/stm32f4/soc.h:27,
12:22:03                  from /home/buildslave/workspace/lite-aeolus-micropython/zephyr/include/arch/arm/cortex_m/cmsis.h:21,
12:22:03                  from /home/buildslave/workspace/lite-aeolus-micropython/zephyr/arch/arm/include/cortex_m/stack.h:27,
12:22:03                  from /home/buildslave/workspace/lite-aeolus-micropython/zephyr/arch/arm/include/kernel_arch_data.h:52,
12:22:03                  from /home/buildslave/workspace/lite-aeolus-micropython/zephyr/kernel/include/kernel_structs.h:65,
12:22:03                  from /home/buildslave/workspace/lite-aeolus-micropython/zephyr/kernel/include/ksched.h:10,
12:22:03                  from /home/buildslave/workspace/lite-aeolus-micropython/zephyr/kernel/timeout.c:10:
12:22:03 /home/buildslave/srv/toolchain/zephyr-sdk-0.9.3/sysroots/armv5-zephyr-eabi/usr/include/stdio.h:227:5: note: previous declaration of 'remove' was here
12:22:03  int _EXFUN(remove, (const char *));
12:22:03      ^~~~~~
~~~

carlescufi, 2018-10-17:
Isn't that a static function though? how come it collides?

andyross, 2018-10-17:
It's a POSIX symbol (an ancient synonym for unlink(), basically), so yeah, sarcasm aside I shouldn't have used it.

Somewhat frustrating that stdio.h (!!!) is getting included out of a HAL layer though.

Will fix.

pfalcon, 2018-10-17:
Even with static function, using simple words like that is not a good practices. Such simple words belong to somebody else.

> how come it collides?

Very simple - a header is included which declares such a function with different prototype. Happens and going to happen, and the only solution is proper namespacing of names.

pfalcon, 2018-10-17:
> Somewhat frustrating that stdio.h (!!!) is getting included out of a HAL layer though.

Any 3rd-party header/lib can be included anywhere (and anytime, mostly of course on the users' side somewhen later).

pfalcon, 2018-10-17:
And don't forget about the debugging case - you're in gdb, and see "remove" function. What does it tell you?

(I would mention map file issue, but static symbols aren't even shown by GUN ld. gdb sees them of course.)

pfalcon, 2018-10-17:
> and see "remove" function

Or "idle", I whatever.

I hope the concern is clear. For networking subsystem there're cleat guidelines on naming: https://github.com/zephyrproject-rtos/zephyr/issues/8723

> Naming convention
.When it is a public function net_&lt;domain><...>()
.When it is a private function &lt;domain><...>() 

And yep, it's surprising that we fight MISRA hydra, but didn't adopted something like above as a golden rule for entire of Zephyr/

andyross, 2018-10-17:
We can go in circles on this all day.  I mean... sure, reusing POSIX symbol names is a bad practice.  But including libc functions seven layers deep out of "hardware abstraction layers" is if anything a worse practice.  If you get to "Any header/lib can be included anywhere" then I get to "any legal symbol can be used", right? :)

Just call out the symbols that bug you and I'll rename them.  Obviously I'll get the POSIX collision fixed.

pfalcon, 2018-10-17:
> sure, reusing POSIX symbol names is a bad practice

But that's not my point, we shouldn't reuse any symbols which can easily belong to somebody else. (Because everyone thinks that they can use it.)

> Just call out the symbols that bug you and I'll rename them. Obviously I'll get the POSIX collision fixed.

My suggestion is to use prefix(es) for all functions of subAPI/subsystem. E.g. timeout_remove(). That's still clash'able, but at least debuggable, set of names.

andyross, 2018-10-17:
That sounds more like a generic complaint about the kernel source, then?  Really I'm asking what it is you want fixed and where.  We have a clearly important build breakage due to a collision on remove(), is that enough to close this bug?

pfalcon, 2018-10-17:
> But including libc functions seven layers deep out of "hardware abstraction layers" is if anything a worse practice.

Of course. But last time I heard, we support out of tree SoCs, boards, drivers, etc. You can imagine what may end up there. So if we do that, let's embrace that it's out of our control, and make users' life easier by containing ourselves in a namespace.

pfalcon, 2018-10-17:
> Really I'm asking what it is you want fixed and where.

I'm trying to raise awareness of the need to namespace things, like I'm trying to do for ~2 years.

> We have a clearly important build breakage due to a collision on remove(), is that enough to close this bug?

Yes, if pledges for more consistent namespacing don't find response in your heart, just fixing specific build breakage is good of course ;-)

pfalcon, 2018-10-17:
I'm not surprised, or CI would fail your PR, right? 3rd-party builds is the key.

Happens here: https://ci.linaro.org/job/lite-aeolus-micropython/548/ . Indeed, it's board-specific, some, but not all ARM boards affected. Build logs can be reached in few clicks, e.g.

https://ci.linaro.org/job/lite-aeolus-micropython/548/PLATFORM=disco_l475_iot1,ZEPHYR_GCC_VARIANT=zephyr,label=docker-xenial-amd64-13/console

pfalcon, 2018-10-17:
(btw qemu_x86 failure there is unrelated)

PR #10670 (2018-10-17): kernel/timeout: Fix build breakage due to stdio name collision
Duh: "remove()" is a POSIX symbol, and on at least some platforms
stdio.h can be included here out of platform headers causing a name
collision.

Fixes #10669's direct issue, though the broader issue of how to choose
names for statics remains controversial.
commit 386894c2 (2018-10-17): kernel/timeout: Fix build breakage due to stdio name collision

Duh: "remove()" is a POSIX symbol, and on at least some platforms
stdio.h can be included here out of platform headers causing a name
collision.

Fixes #10669's direct issue, though the broader issue of how to choose
names for statics remains controversial.
closed 2018-10-17 by nashif