zephyrproject-rtos/zephyr #10671: Callback function le_param_req() is never called.

SergeSmirnov, 2018-10-17:
A simplest example is samples\bluetooth\peripheral_dis.

Board nrf51_pca10028.

Minor changes to the main.c file.
Functions le_param_req() and le_param_updated() added to conn_callbacks{} structure. 
```
static bool le_param_req(struct bt_conn *conn,                        ////
          struct bt_le_conn_param *param)                             //
{                                                                     //
  printk("le_param_req\n");                                           //
  return true;                                                        //
}                                                                     // added lines
                                                                      //
static void le_param_updated(struct bt_conn *conn, u16_t interval,    //
    u16_t latency, u16_t timeout)                                     //
{                                                                     //
  printk("le_param_updated\n");                                       //
}                                                                     ////

static struct bt_conn_cb conn_callbacks = {
  .connected = connected,
  .disconnected = disconnected,
  .le_param_req     = le_param_req,                                   // added line
  .le_param_updated = le_param_updated,                               // added line
};
```
For the test, nRF connect PC is used. Three simple steps:
1. Connection
2. Change connection parameter
3. Disconnection

Log:
```
***** Booting Zephyr OS 1.13.99 *****
Bluetooth initialized
Advertising successfully started
Connected
le_param_updated
le_param_updated
Disconnected (reason 19)
```

It can be seen that the parameters have changed, there is call le_param_updated(), but there is no call le_param_req().

jhedberg, 2018-10-18:
IIRC the central can just go and change the parameters - it does not need the peripheral's permission for it. Probably that's what's happening. The le_param_req() callback is in that sense mostly useful for central applications and not peripheral ones. @sjanc @carlescufi please correct me if I'm wrong.

SergeSmirnov, 2018-10-18:
For example. The peripheral device at the connection set latency 4 and does not want to change it. 
And then the central device requested a latency 0. How can a peripheral device refuse to change parameters if it does not know about it.

carlescufi, 2018-10-19:
@SergeSmirnov there are two ways of changing connection parameters in BLE. I will use the Link Layer names:

* LE Connection Update: This was introduced in BLE 4.0. Only the master can make use of it and it can change the connection parameters at any time without the slave being able to do anything about it. However the slave can request connection parameters through a "hint" packet that it can send over L2CAP (Connection Parameter Update Request) to request the master to change the actual connection parameters via an LE Connection Update.

* LE Connection Parameters Request: This was introduced in BLE 4.1. This can be initiated by both master and slave, and both can reject the parameters if they don't accept their values.

I believe your master (nRF Connect on PC) is using "LE Connection Update" to change the connection parameters unilaterally without consulting the slave. There is not much you can do about that except to request another change of connection parameters after using an L2CAP Connection Parameter Update Request.
You can enable the LE Connection Parameters Request feature in the controller by setting `CONFIG_BT_CTLR_CONN_PARAM_REQ=y`.

SergeSmirnov, 2018-10-19:
This is very useful information, thanks.
Unfortunately, adding CONFIG_BT_CTLR_CONN_PARAM_REQ=y has not changed anything.

What other  test can I do to distinguish the master uses **LE Connection Update** or **LE Connection Parameters Request**?
Or which other software master uses **LE Connection Parameters Request** to test the function le_param_req()?

carlescufi, 2018-10-30:
@SergeSmirnov so there are 2 things here:

1) Are you sure `CONFIG_BT_CTLR_CONN_PARAM_REQ` is set to `=y` in your `<build>/zephyr/.config`?
2) You need to be sure that the master (nRF_Connect) actually supports LE Connection Parameters Request, because both master and slave need to support it in order for the feature to be used.

@cvinayak any other clues?

carlescufi, 2018-10-30:
@SergeSmirnov also the decision is taken here: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/conn.c#L204

Maybe you can add some debug info there to find out exactly why L2CAP is being used?

SergeSmirnov, 2018-10-30:
@carlescufi
About **CONFIG_BT_CTLR_CONN_PARAM_REQ** just double checked it.
By default **CONFIG_BT_CTLR_CONN_PARAM_REQ=y**. 
To disable it you need to set **CONFIG_BT_CTLR_CONN_PARAM_REQ=n**.

Thanks for the link. I will try debugging tomorrow.

SergeSmirnov, 2018-10-31:
@carlescufi
First, a big thank you for the clue the direction of the search.
I did a lot of tests with various Android and Apple devices using nRF_Connect.
Checked commands LL_FEATURE_REQ, LL_SLAVE_FEATURE_REQ, LL_FEATURE_RSP and LL_VERSION_IND.

If the BLE version of the master is 4.2, then feature **Connection Parameters Request Procedure** is allowed and the callback **le_param_req()** occurs.

nRF_Connect PC latest version for today (2.5) reports BLE version 4.0.
Interestingly, if you install a new version of the OS on old devices, then BLE version 4.0 remains, new features do not appear. 
For example, the phone had a version of Android 4, now Android 7. Ipad had a version of iOS 6, now IOS 10.

If you set **CONFIG_BT_CTLR_CONN_PARAM_REQ=n** when working with a device with BLE version 4.2, then everything works as with BLE version 4.0.

Log with debug line from your link:
1. BLE 4.0
```
***** Booting Zephyr OS 1.13.99 *****
Bluetooth initialized
Advertising successfully started
Connected
le_param_updated
le_param_updated
conn 0x20000508 features 0x00 params (24-40 0 42)
le_param_updated
```

2. BLE 4.2
```
***** Booting Zephyr OS 1.13.99 *****
Bluetooth initialized
Advertising successfully started
Connected
le_param_req
le_param_updated
le_param_req
le_param_updated
conn 0x20000508 features 0xff params (24-40 0 42)
le_param_updated
```

So my bug report is wrong.

carlescufi, 2018-10-31:
@SergeSmirnov thanks very much for reporting back. I will now close this issue, glad that you got your issue resolved and look forward to you using Zephyr for your applications.

closed 2018-10-31 by carlescufi