zephyrproject-rtos/zephyr #10672: nRF52: UARTE: uart_fifo_fill return value unreliable

tautologyclub, 2018-10-17:
Today I noticed that when using interrupt-driven UART (UARTE @ UART_1 on a nrf52840 based board), calling `uart_fifo_fill(buf_ptr, buf_sz)` often returned `buf_sz` despite not pushing the whole buffer onto the fifo. Speficially, this always occured for me when `buf_sz > 8` and never when `buf_sz < 5` (not sure what happens in between but I guess 8 is a reasonable guess at a cut-off point).

I.e. I'd try to push 16 bytes onto the fifo from an ISR context, `uart_fifo_fill` returns 16, with no error reported at any point, but only part of the buffer gets shipped to the UART. 

I guess the hardware buffer size is actually eight bytes, but I couldn't find it documented anywhere -- in any case, I'd say the lack of error/assertion/indication would classify this as a bug.

carlescufi, 2018-12-13:
@jarz-nordic could you please take a look at this issue?

jarz-nordic, 2018-12-13:
@tautologyclub do you use the newest master? Do you disable interrupts before tx ready is indicated?

jarz-nordic, 2018-12-13:
@carlescufi @tautologyclub : It was reported 17th of October, I think we already fixed it recently.
Please let me know if you still observe this issue.

tautologyclub, 2018-12-19:
I disabled tx interrupt: 

```
uart_irq_tx_disable(dev);
ret = uart_fifo_fill(dev, tx_queue, sz);
```

I'll try to rebase and check when I have time. You don't happen to have a commit reference on hand, right?

jarz-nordic, 2018-12-19:
I see two problems here. First of all you shall not call `uart_fifo_fill` once interrupts are disabled - uart_irq_tx_ready funciton returns false. It is against UART API.

You need to change order to:
```
ret = uart_fifo_fill(dev, tx_queue, sz);
uart_irq_tx_disable(dev);
```
Next we have had a problem in our driver that calling `uart_irq_tx_disable(dev);` immediately disables UARTE transmitter. As a result you will observe that not all bytes have been shifted out. We have fixed this problem in PR: #11475.

So if you will rebase to current master and you will change order of function calls i you should no longer have a problem.

tautologyclub, 2018-12-19:
I'll try to pull the changes. 

Btw, here is a more complete snippet:

```
while (uart_irq_update(dev) && uart_irq_is_pending(dev)) {
	if (uart_irq_tx_ready(dev)) { 
		uart_irq_tx_disable(dev);
		ret = uart_fifo_fill(dev, tx_queue.buffer + tx_queue.head,
				     fifo_bytes);
		...
	}
```

But you're saying that this is incorrect, but why? Is it because Nordics implementation of `uart_fifo_fill` uses `uart_irq_tx_ready()` internally and depends upon it returning true, or is it "always" wrong? 
I tried to follow the docs in `include/uart.h` for `uart_fifo_fill`, but they aren't very clear here. It only states you need to call the function from isr context, not that the tx interrupt needs to be enabled.

jarz-nordic, 2018-12-19:
I bet that our UART and UARTE driver can behave in different way in your scenario. It is very likely that UART will work because transmitter is not switched off on `uart_irq_tx_disable`. This feature has been implemented only for UARTE to increase power efficiency according to chip specification.

tautologyclub, 2018-12-19:
I see your point re: `uart_irq_tx_ready`.

Btw, I'm actually using UARTE (UART_1, interrupt-driven). Interesting to note is that the upstreamed `wncm14a2a.c` driver calls `uart_fifo_fill` with seemingly reckless abandon, and that modem is included in the nrf52840_pca10056 dts. Weird.

mike-scott, 2018-12-20:
@tautologyclub Ugh, you are correct that 'uart_fifo_fill' was sloppily used in the modem driver.

It was changed from 'poll_out' implementation at the last minute and it looks like it's broken on nRF52840.  Oddly, it works on NXP and STM boards. :/.

tautologyclub, 2018-12-28:
But I can't believe the Zephyr tree contains no working samples of interrupt-driven UART handlers that doesn't break the `uart_fifo_fill` API right...? What's an example?

closed 2019-01-24 by tautologyclub