zephyrproject-rtos/zephyr #10676: Feature Required: DFU over Thread network

arn-soft, 2018-10-18:
It would be great, if we can have DFU over thread network feature in zephyr or may be a network agnostic dfu system.

NMzen, 2020-06-11:
Hi Nashif
Any update on DFU over thread network in zephyr? Has it been accomplished or still awaited?

Thanks, NM

rlubos, 2021-07-08:
Thread specification per se does not specify any particular DFU mechanism, so deploying DFU over Thread network is rather a matter of running any of the existing network-agnostic DFU protocols over Thread. You could use for instance LwM2M with its Firmware Update object, updatehub or hawkbit. But there's no out-of-the box example demonstrating how to combine these DFU protocols with Thread. 

Please also see https://github.com/zephyrproject-rtos/zephyr/issues/29740, which covered the same topic.

closed 2021-07-08 by nashif