zephyrproject-rtos/zephyr #10678: nRF52840: hci_usb timeout during initialization by bluez

Eystorhaf, 2018-10-18:
Hey there,

encountered a problem with hci_usb sample on nrF52840,

Hardware:
USB-Devices:
nordic nRF52840 DK & laird DVK-BL654-1.0

USB-Host:
PC / x86 Ubuntu LTS 18.04 / Bluez 4.48 / Kernel 4.15.0-36-generic

Firmware:
zephyr-master @8b20f9d  

While initializing the hci-device bluez gets a timeout:
[Thu Oct 18 02:36:18 2018] Bluetooth: hci0: command 0x1002 tx timeout 

Here the output of:

btmon: https://pastebin.com/ypNTmTgC
journalct -f | grep bluetoothd* :https://pastebin.com/PT4V6Dpf
dmesg -T:  https://pastebin.com/VQQ0tf1x
zephyr uart debug console: https://pastebin.com/v14JUCux

Wondering whether this is a usb or hci stack problem?

Best,

jhedberg, 2018-10-18:
I don't know why the controller would not respond to Read Local Supported Commands. It's something that's used even as part of a combined build HCI initialization.

jhedberg, 2018-10-18:
The only special think I can think of regarding HCI_Read_Local_Supported_Commands is that its return parameters (as part of the Command Complete event) are probably some of the longest of any HCI command. Perhaps that could be an issue here (possibly with the way the HCI transport is implemented).

cvinayak, 2018-10-19:
@jhedberg @uhrheber I just now built the hci_usb sample, and it works out-of-the-box as a btusb device on my archlinux (on my virtualbox) when flashed on my nRF52840 DK.

pawelzadrozniak, 2018-10-19:
The device does not enumerate on Windows. Looks like HCI ACL packet is interpreted incorrectly:
https://pastebin.com/0BUYatVF
Error in hci.c (added length logging):
`	if (buf->len < len) {
		BT_ERR("Invalid HCI ACL packet length, %d < %d -", buf->len, len);
		return -EINVAL;
	}`

17218 equals 0x4342 - see the attached USB log:
[image]

pawelzadrozniak, 2018-10-19:
Looks like Windows 7 does not support this type of device.
The device enumerates in Linux, however I am getting bus fault BT thread:
https://pastebin.com/ffugF07U

henrikbrixandersen, 2018-10-22:
I can confirm this issue. I tried reverting 68cb7b3 and related commits - that works.

pawelzadrozniak, 2018-10-30:
Timeout issue fixed in #10934 .

henrikbrixandersen, 2018-11-10:
I can confirm that the timeout issue was fixed in #10934. Can this issue be closed?

cvinayak, 2018-11-12:
Fixed by #10934 

Thank you @henrikbrixandersen

closed 2018-11-12 by cvinayak