zephyrproject-rtos/zephyr #10693: samples/subsys/usb/console: Console routed to UART of no USB

erwango, 2018-10-18:
When testing samples/subsys/usb/console, "Hello World" is routed to UART console if USB console is not there.
I don't know if this is intended behavior, but this could be confusing when testing.

zerotri, 2018-10-22:
This totally caught me off guard as well a week ago when I was trying to get a usb shell working on stm32l4. We could maybe add a disclaimer in the README for this sample.

finikorg, 2018-11-21:
So what is the problem? Missing console?

Since we change console parameters to DTS we have to for every board include the same DTS overlay. Just copy existing overlay to your board name :)

finikorg, 2018-11-21:
Maybe if DTS magic does not include USB console we could issue #error so that user notice that he needs to enable it?

PR #11559 (2018-11-21): usb: console: Notify user about incorrect configuration
Since setting USB console port name is moved to DTS users have
problems enabling it. Notify user that the console is not set.

Fixes #10693
commit 971be355 (2018-11-21): usb: console: Notify user about incorrect configuration

Since setting USB console port name is moved to DTS users have
problems enabling it. Notify user that the console is not set.

Fixes #10693
closed 2018-11-21 by nashif