zephyrproject-rtos/zephyr #10701: API: Prefix (aio_) conflict between POSIX AsyncIO and Designware AnalogIO Comparator

pfalcon, 2018-10-18:
Following the lovely topic of https://github.com/zephyrproject-rtos/zephyr/issues/10669, etc. I just spotted not a direct name clash (yet), but prefix clash. So, POSIX defines async i/o sub-api with functions starting with `aio_`, e.g. `aio_` http://pubs.opengroup.org/onlinepubs/9699919799/basedefs/aio.h.html . 

Surely, sooner or later we'll get to those (I remember @d3zd3z already asked if we should implement them).

But turned out, we have https://github.com/zephyrproject-rtos/zephyr/blob/master/include/aio_comparator.h with a number of functions starting with the same prefix. To objective, it's more like `aio_cmp_`, but still close enough and can be confusing.

As we can't change POSIX, we'd rather rename that API, e.g. to `ana_cmp_`.

pfalcon, 2018-10-18:
@nashif : FYI

d3zd3z, 2018-10-19:
If anything, renaming "AnalogIO" to something other than `aio_` will help avoid confusion. AIO is fairly commonly used to refer to asynchronous I/O, and not very commonly used to refer to Analog IO.

A good rule to use when trying to come up with an acronym is to do a Google search for the acronym. If your meaning isn't near the top, and especially if there are already other meanings, pick something else.

pfalcon, 2019-10-03:
Related PR https://github.com/zephyrproject-rtos/zephyr/pull/16114 was closed with a notice "aio drivers and header removed, closing...", so it makes sense to close this too. But whoever may want to reuse aio_ prefix for something else, please google up this issue and leave it reserved for POSIX!

closed 2019-10-03 by pfalcon