zephyrproject-rtos/zephyr #10703: net: possible "disable" flag for the bulk of network traffic classification

mike-scott, 2018-10-18:
The RAM usage alone for net_tc is 3.5K.   That is very valuable space atm.

Many applications may not need this, is there a way we could device a flag to disable it?

jukkar, 2018-10-19:
The net_tc.c is implementing the same things as what we did earlier in net_if.c and net_core.c, see also commit 6049207a29a3a74b238571470336a87f3372d0d7 that introduced net_tc.c.
By default there is only one traffic class which implements the same functionality that we had in net_if.c (for TX) and in net_core.c (for RX). Normally you should only have one traffic class, this is the default. In this case the RAM consumption should be about the same as what we had before.

Just to prove this memory consumption, I did some experiment by calculating RAM consumption before and after the commit I mentioned in first sentence. This is the echo-server sample in qemu_x86_nommu board.

```
Before net_tc.c introduction:

net_core.c                        1648     4.61%
  __config_sys_init_net_init1       12     0.03%
  __device_sys_init_net_init1       12     0.03%
  __net_stack_RX_rx_stack_0         28     0.08%
  rx_queue                          16     0.04%
  rx_stack                        1500     4.20%
  rx_thread_data                    56     0.16%
  startup_sync                      24     0.07%

net_if.c                          2200     6.15%
  __net_stack_TX_tx_stack_0         28     0.08%
  ipv4_addresses                   152     0.43%
  ipv6_addresses                   612     1.71%
  link_callbacks                     8     0.02%
  mcast_monitor_callbacks            8     0.02%
  routers                          136     0.38%
  tx_stack                        1200     3.36%
  tx_thread_data                    56     0.16%
                                  ====
                                  3848
```

```
After net_tc.c introduction:

net_tc.c                          2916     8.04%
  __net_stack_RX_rx_stack_0         28     0.08%
  __net_stack_TX_tx_stack_0         28     0.08%
  rx_classes                        80     0.22%
  rx_stack                        1500     4.13%
  tx_classes                        80     0.22%
  tx_stack                        1200     3.31%

net_if.c                           916     2.52%
  ipv4_addresses                   152     0.42%
  ipv6_addresses                   612     1.69%
  link_callbacks                     8     0.02%
  mcast_monitor_callbacks            8     0.02%
  routers                          136     0.37%

net_core.c                          24     0.07%
  __config_sys_init_net_init0       12     0.03%
  __device_sys_init_net_init0       12     0.03%
                                  ====
                                  3856
```

The difference is 8 bytes here, not sure why is that but I think we can live with this.

mike-scott, 2018-10-19:
@jukkar Heh, ok that's a very complete analysis.

My misperception was that TC logic added code / resources which wouldn't be needed if we only had 1 classification.

I'll close.

closed 2018-10-19 by mike-scott