zephyrproject-rtos/zephyr #10727: Documentation: Need better documentation

dleach02, 2018-10-19:
The documentation on how to setup debugging is not very clear and the existing documentation is spread around in a difficult to grep manner. For example, the NXP boards have some fairly decent documentation but when I switched to debugging on the Reel Board it would not work in my environment. Turns out I needed to add a udev rule which was buried pretty deep in the documentation tree for the 96boards and this was only pointed out after discussions with another person who remembered seeing this.

The documentation is a bit vague and spread on how to setup debug environments. My suggestion is to pull up debug setup to a higher level and deal with the various ways we currently debug Zephyr.

qianfan-Zhao, 2018-10-19:
+1 should add debugger tool's usb device id to udev and avoid use root debug and download.

But seems there are no such documents or it is hard to find in the documents. I suggest that add them to 'Development Environment Setup on Linux'

nashif, 2019-02-19:
@dleach02 how about you take this? :-)

carlescufi, 2021-02-25:
Make sure to document `addr2line` as part of this issue.

dleach02, 2021-04-01:
@carlescufi are you now asking/thinking about a gdb cheat sheet (which there are many) for debugging Zephyr applications from gdb CI?

carlescufi, 2021-04-26:
> @carlescufi are you now asking/thinking about a gdb cheat sheet (which there are many) for debugging Zephyr applications from gdb CI?

No, I was really just thinking about `addrline`, which is not part of gdb but is very useful when trying to trace back the source of an assert or fault.

dleach02, 2021-04-26:
I've used 'info line *<addr>' from within gdb or 'info symbol <addr>' ... haven't used addr2line

closed 2025-11-16 by dleach02