zephyrproject-rtos/zephyr #10748: Work waiting on pollable objects

pdunaj, 2018-10-22:
At the moment Zephyr has two kinds of works - executed straightaway or executed with delay. It would be useful to have a third kind in which work would be executed when specific pollable object is ready (or any from a given list).

This way one could avoid creating a new thread to wait for some blocking API to become ready. Since creation of a thread costs RAM it may be to expensive to create many threads for every small task that use blocking read or something similar.

This could be solved by creating a dedicated thread that waits for any of the registered objects and when ready submit an associated work.
Or this could be hooked right into the scheduler - when object is ready and there is any work associated with it, such work would be submitted for execution.

pfalcon, 2018-10-22:
Somehow, "This way one could avoid creating a new thread" and "This could be solved by creating a dedicated thread" together don't ring well to me.

But anyway, if you think it's useful for your needs, why don't you try to prototype it, and then see if it's really useful to become a generic facility of kernel (that's my understanding is what you propose; and that likely would require major changes to the "prototype" to turn it into fully general and bug-free facility).

> Or this could be hooked right into the scheduler

No please. We have had enough bugs in the scheduler, and still have more, so loading it up with more arguably adhoc features doesn't sound cheerful at all.

pdunaj, 2018-10-23:
Hi @pfalcon 

> together don't ring well to me.

If you need to block in one place it would be no problem. If you need to block in 10 places it is a problem and this is what I refer to.

> why don't you try to prototype it

I will once I have a bit of time. Maybe somebody else will also fine the idea useful and do it earlier.

closed 2019-10-04 by carlescufi