zephyrproject-rtos/zephyr #10755: [Coverity CID :188885]Error handling issues in /lib/ring_buffer/ring_buffer.c

mandarcthorat1, 2018-10-22:
Static code scan issues seen in File: /lib/ring_buffer/ring_buffer.c
 Category: Error handling issues
 Function: ring_buf_put
 Component: Other
 CID: 188885
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

ceolin, 2018-10-26:
It's not an issue. The return can be ignored because the only possible error is covered by `ring_buf_put_claim()`. Though, it is always a good practice checking the return even if it is just a cast to explicit acknowledge that it can be ignore.

mandarcthorat1, 2019-03-12:
This coverity defect has been marked as fixed in latest scan as the file path has changed since issue creation. Pull request #14150 is still relevant.

PR #14150 (2019-03-14): lib: os: ring_buffer: Fix not handled return value
Some function return values were not handled. Added assert in case
those functions return error. It is possible only if same ring buffer
instance is used without any protection from multiple contexts.

Fixes #10755.
closed 2019-03-14 by nashif