zephyrproject-rtos/zephyr #10760: Flash Shell compilation error without BT

derchrismakaio, 2018-10-22:
Compiling without BT shell I get a compilation error for line
extern const struct shell *ctx_shell;

If I change to
const struct shell *ctx_shell;

it works, so I guess w/o BT shell it's simply not exposed.
However, I wonder why this is needed as the shell comes as param in every cmd

Vudentz, 2018-10-29:
I wonder what the flash has to do with BT shell here? Note that any asynchronous operation that requires to print out something to the shell need a shell pointer in order to do so, which is what the ctx_shell is used for.

derchrismakaio, 2018-11-21:
@Vudentz well, my own application... I'll keep an eye on this and when I stumble accross a sample which gives the same error I'll let you know

galak, 2018-12-13:
Going to close this, please re-open if you can provide a reproducible test case showing the issue.

vanwinkeljan, 2019-01-08:
Problem is reproducible by adding following configs to hello_world sample prj.conf and building for reel_board:
```
CONFIG_FLASH=y
CONFIG_FLASH_SHELL=y
```

vanwinkeljan, 2019-01-09:
fix is merged

closed 2019-01-09 by vanwinkeljan