zephyrproject-rtos/zephyr #10775: Build error in zephyr.git/include/toolchain/gcc.h leading to shippable failures

spoorthik, 2018-10-23:
Log:

```
FAILED: ccache /usr/bin/gcc -DBUILD_VERSION=zephyr-v1.13.0-1239-g5721c3b1d -DKERNEL -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -D__ZEPHYR__=1 -I../../../../../../kernel/include -I../../../../../../arch/posix/include -I../../../../../../soc/posix/inf_clock -I../../../../../../soc/posix/inf_clock/include -I../../../../../../soc/posix/include -I../../../../../../boards/posix/native_posix -I../../../../../../include -I../../../../../../include/drivers -Izephyr/include/generated -I../../../../../../ext/lib/crypto/tinycrypt/include -I../../../../../../subsys/bluetooth -I../../../../../../tests/include -O0 -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/out-2nd-pass/native_posix/tests/bluetooth/shell/test/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -Werror -Wno-deprecated-declarations -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -Wpointer-arith -ffunction-sections -fdata-sections -fno-freestanding -m32 -MMD -MP -include /home/buildslave/src/github.com/zephyrproject-rtos/zephyr/arch/posix/include/posix_cheats.h -fprofile-arcs -ftest-coverage -std=c11 -MD -MT zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj -MF zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj   -c zephyr/misc/generated/configs.c
In file included from ../../../../../../include/toolchain.h:21:0,
                 from zephyr/misc/generated/configs.c:3:
zephyr/misc/generated/configs.c: In function ‘_ConfigAbsSyms’:
../../../../../../include/toolchain/gcc.h:362:48: error: expected expression before ‘)’ token
   "\n\t.type\t" #name ",@object" :  : "n"(value))
                                                ^
zephyr/misc/generated/configs.c:196:1: note: in expansion of macro ‘GEN_ABSOLUTE_SYM’
 GEN_ABSOLUTE_SYM(CONFIG_NET_BUF_LOG_LEVEL, );
 ^~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```

aescolar, 2018-10-23:
@SebastianBoe : It seems to be an issue with the autogenerated misc/generated/configs.c being in some cases corrupted. Do you have some clue about why is this happening?

aescolar, 2018-10-23:
@jukkar All errors seem to relate to `CONFIG_NET_BUF_LOG_LEVEL`

SebastianBoe, 2018-10-23:
Has anyone been able to reproduce this locally?

It would be interesting to see what config.c looks like.

aescolar, 2018-10-23:
@SebastianBoe : At least I wan't able to reproduce it locally. But  from the error in CI:
`GEN_ABSOLUTE_SYM(CONFIG_NET_BUF_LOG_LEVEL, ):`

jukkar, 2018-10-23:
This is already fixed by #10763

jukkar, 2018-10-23:
The issue was seen if one enabled CONFIG_NET_BUF_LOG=y, which happened for couple of Bluetooth tests.

closed 2018-10-23 by jukkar