zephyrproject-rtos/zephyr #10779: ext: NULL dereferences

himanshujha199640, 2018-10-23:
Found using Coccinelle.
https://github.com/zephyrproject-rtos/zephyr/pull/10776
```
himanshu@himanshu-Vostro-3559:~/zephyr$ ./scripts/coccicheck --mode=report --cocci=scripts/coccinelle/deref_null.cocci --verbose=1 ext/ 
Please check for false positives in the output before submitting a patch.
When using "patch" mode, carefully review the patch before submitting it.

Processing deref_null.cocci
with option(s) ""

Message example to submit a patch:

 A variable is dereferenced under a NULL test.
 Even though it is known to be NULL.

 The semantic patch that makes this report is available
 in scripts/coccinelle/deref_null.cocci.

 More information about semantic patching is available at
 http://coccinelle.lip6.fr/

Running (4 in parallel): /usr/local/bin/spatch -D report --no-show-diff --very-quiet --cocci-file scripts/coccinelle/deref_null.cocci --patch /home/himanshu/zephyr --dir ext/ --jobs 4 --chunksize 1
ext/hal/st/stm32cube/stm32l4xx/drivers/src/stm32l4xx_hal_i2c.c:604:10-19: ERROR: hi2c is NULL but dereferenced.
ext/hal/st/stm32cube/stm32l4xx/drivers/src/stm32l4xx_hal_i2c.c:494:10-19: ERROR: hi2c is NULL but dereferenced.
ext/lib/ipc/open-amp/open-amp/lib/remoteproc/drivers/zynqmp_remoteproc_a53.c:224:17-22: ERROR: ipi is NULL but dereferenced.
ext/lib/ipc/open-amp/open-amp/lib/remoteproc/drivers/zynqmp_remoteproc_r5.c:182:17-22: ERROR: ipi is NULL but dereferenced.
```

erwango, 2018-10-23:
For, stm32cube, NULL check is performed to detect handle initialisation issue, and return error is true.
So, we can argue this should not happen, but this should be considered as a protection on the use of the API (function should not be called if handle has not been initialized first) made on purpose and not an implementation issue.

himanshujha199640, 2018-10-23:
> For, stm32cube, NULL check is performed to detect handle initialisation issue, and return error is true.
> So, we can argue this should not happen, but this should be considered as a protection on the [...]

Do you mean NULL dereference is made on purpose ?
That is actually an underfined behavior according to C Standard and compiler can do **anything** under
such circumstance.

On the other hand, if you say that "Yes! That condition shouldn't happen in any case and let's just nuke it ..." knowing that it could lead to a kernel Oop.

I don't know much about the code in discussion. So if it is a false positive, then feel free to close(or mark resolved) this issue.

erwango, 2018-10-24:
Checked code again and while code is made to detect NULL pointer and return an error, it is actually doing NULL pointer dereference in between...
I will raise an ST internal issue and will get back here to provide bug tracker ID so we can track this issue.

erwango, 2018-10-29:
STM32Cube issue has been fixed in upcoming version of the CubeL4 package (V1.13.0, current V1.12.0).
Will be fixed by: #10642

PR #10642 (2018-10-29): ext/hal: Cube update STM32L4 and F7 fixup
STM32L4 was missing from last cube update session.
Provide latest version and fix STM32F7 README.

Fixes #10779
closed 2018-10-29 by galak