zephyrproject-rtos/zephyr #10801: Compiler warning in soc/xtensa/esp32/include/_soc_inthandlers.h

spoorthik, 2018-10-24:
Commit: 7cea94942c35bb65a8c219b69ec2270c0fedd6f4

Log:

```
In file included from /home/kspoorth/work/forked_zephyr/arch/xtensa/core/xtensa-asm2.c:14:0:
/home/kspoorth/work/forked_zephyr/soc/xtensa/esp32/include/_soc_inthandlers.h: In function '_xtensa_handle_one_int0':
/home/kspoorth/work/forked_zephyr/soc/xtensa/esp32/include/_soc_inthandlers.h:363:1: warning: label 'handle_irq' defined but not used [-Wunused-label]
 handle_irq:
 ^
/home/kspoorth/work/forked_zephyr/soc/xtensa/esp32/include/_soc_inthandlers.h: In function '_xtensa_handle_one_int6':
/home/kspoorth/work/forked_zephyr/soc/xtensa/esp32/include/_soc_inthandlers.h:373:1: warning: label 'handle_irq' defined but not used [-Wunused-label]
 handle_irq:
 ^
```

It is good to get rid of this as it would be an issue while building with compiler warnings being treated as errors.

nniranjhana, 2018-10-25:
This is rather a priority since many tests enabled on ESP32 are failing on build now.

This is the code causing the issue -

```
static inline int _xtensa_handle_one_int0(unsigned int mask)
{
        int irq;

        return 0;
handle_irq:
        _sw_isr_table[irq].isr(_sw_isr_table[irq].arg);
        return mask;
}
```
How would one reach the label `handle_irq` from inside this function tho?

spoorthik, 2018-10-25:
FYI: @lpereira

commit aa442d13 (2018-11-06): arch: xtensa: Do not generate unused `handle_irq` label unless needed

This was causing a `label handle_irq defined but not used` warning
during build.

Fixes #10801.
commit d3528cf0 (2018-11-06): soc: esp32: Update auto-generated _soc_inthandlers.h

Remove the auto-generated `handle_irq` label for the unneeded cases
to get rid of a compile warning.

Fixes #10801.