zephyrproject-rtos/zephyr #10811: crash while connecting a USB cable

pdunaj, 2018-10-24:
I have enabled USB HID class. On my device almost everything seems to be working fine except information about USB cable connection (for some reason it seems to be always attached, even if it is not). I suspect a HW bug so I took a second device. This time I did not get a notification about connection while cable is disconnected. When the USB was attached however I got following error:
`ASSERTION FAIL [0] @ /home/padu/work/fw-nrfconnect-zephyr/drivers/clock_control/nrf5_power_clock.c:322`

It seems that low frequency clock has not started.

pdunaj, 2018-10-24:
In `_power_clock_isr`, when USB is enabled, there are three more: `usb_detected`, `usb_power_rdy` and `usb_removed`. This three seems to be connected with what I was doing (crash happened when I connected the USB).

We use qdec which as far as I remember depends on low frequency clock. Then it won't be surprising that lf is true (was started). Why crashing then?

pdunaj, 2018-10-24:
@kl-cruz , @pawelzadrozniak do you have any ideas?

pdunaj, 2018-10-24:
on the device that "works" I get following sequence (note that USB cable is not connected):
These three int are the usb related variables in the order they are set.
```
***** Booting Zephyr OS v1.13.99 *****
power usb: 1 0 0 lf:0
attached
power usb: 0 1 0 lf:0
powered
USB connected
```

On the failing device I get following:
```
***** Booting Zephyr OS v1.13.99-ncs1-822-gcd2c76f *****
//Here I connect USB cable
power usb: 1 0 0 lf:1
ASSERTION FAIL [0] @ /home/padu/work/fw-nrfconnect-zephyr/drivers/clock_control/nrf5_power_clock.c:323:
```

When cable is connect from the start (before device is reset) there is no crash. Sequence is like in the first case.

pdunaj, 2018-10-24:
When I comment out this assert sometimes I get the event about USB being detached and sometimes I don't... It doesn't seem very reliable...

pdunaj, 2018-10-25:
Hi @cvinayak , I had a look into the clock driver code and I think the problem is caused by the fact that when `CONFIG_CLOCK_CONTROL_NRF5_K32SRC_BLOCKING` is not selected LF clock is started but status is never confirmed. When USB cable is attached `lf` reads out information that at some point in time low frequency clock was set. This should be fixed - either by removing an assert in `_power_clock_isr` or by somehow tracking that we are asynchronously waiting for LF clock to be settled.

cvinayak, 2018-11-03:
@pdunaj yes #11038 will fix the accidental assert due to LF event being set when USB events are enabled.

pdunaj, 2018-11-05:
@cvinayak that is a good news thanks! Could you make the PR close the issue once it is merged?

cvinayak, 2018-11-05:
@pdunaj I guess having a "Fixes ..." in commit message should close this issue, hopefully, else will close is manually.

carlescufi, 2018-11-05:
Fixed in master

commit ebaabba8 (2018-11-05): clock_control: nrf5: Ensure HF & LF running before calibration

Fix RCOSC calibration initiation to ensure LF clock is
running; and the HF clock is running with crystal before
triggering the actual calibration in ISR.

Fixes #10811.
closed 2018-11-05 by carlescufi