zephyrproject-rtos/zephyr #10821: ELCE: DT, Kconfig, EDTS path forward

carlescufi, 2018-10-24:
- Run DT before Kconfig
  - Remove `#if CONFIG_` from all `board/` and `dts/` files
  - Change build system so that DT runs before Kconfig

- Change Kconfig symbols so that they are in synch with driver enablement
  - Generate a fragment file from DT data so that a `compat=spi-stm32` generates `CONFIG_SPI_STM32`
  - `CONFIG_SPI_STM32` selects `CONFIG_SPI`

- Remove `#if CONFIG_*`

galak, 2018-10-24:
Here are the defines we need to remove board/*/*.dts and dts/

```
CONFIG_BOARD_EM_STARTERKIT_R23
CONFIG_BOOTLOADER_MCUBOOT
CONFIG_BT
CONFIG_CODE_HYPERFLASH
CONFIG_CODE_ITCM
CONFIG_CODE_QSPI
CONFIG_FS_FLASH_MAP_STORAGE
CONFIG_FS_FLASH_STORAGE_PARTITION
CONFIG_MCUMGR_SMP_UART
CONFIG_MODEM_WNCM14A2A
CONFIG_SOC_EMSK_EM11D
CONFIG_SOC_EMSK_EM7D
CONFIG_SOC_EMSK_EM9D
CONFIG_UART_1_NRF_UARTE
CONFIG_UART_MCUX_2
CONFIG_USB_UART_CONSOLE
CONFIG_XIP
```

carlescufi, 2018-10-24:
> > CONFIG_SPI_STM32 selects CONFIG_SPI
> 
> This would mean that we enable drivers though DT, we don't want to do that do we?

Sorry we were just brainstorming, see the updated description.

SebastianBoe, 2018-10-25:
I believe that this is what will require the most work and break compatibility the hardest:

> Remove #if CONFIG_ from all board/ and dts/ files

the rest looks fairly straightforward.

I would like to discuss our strategies for getting these Kconfig symbols out of DT. For instance CONFIG_BOOTLOADER_MCUBOOT. I agree with the assessment of @mbolivar , who has worked on the MCUboot integration, that taking the MCUBoot partitions out of the scope of DT is the best approach here[0].

I am aware that communicating flash partitioning layout information between the bootloader and the kernel is one of responsibilities of DT. But I believe that although this has worked well for other DT users, it does not work well for us.

[0] https://github.com/zephyrproject-rtos/zephyr/pull/9925#discussion_r216826127

Can we agree on this, or are there any viable alternatives?

carlescufi, 2018-11-08:
> @carlescufi :
> 
> > Have DT generate Kconfig fragment enabling the compatibles: CONFIG_DT_COMPAT_STM32_SPI=y so that if the user enables CONFIG_SPI=y then the right driver is enabled.
> 
> About that point: you mentioned a reason why COMPAT flags should remain CONFIG_DT_XXX since having them in DT_ namespace would not work (possible but would not enable the expected usage). Can you state here the reason again?

@erwango 
I believe it's because we are generating a Kconfig fragment, and Kconfig variables **must** be prefixed with `CONFIG_`

erwango, 2018-11-08:
@carlescufi , ok so should be excluded from https://github.com/zephyrproject-rtos/zephyr/pull/11180 then

carlescufi, 2020-02-27:
This is not really relevant anymore, and the missing bits are covered by #8499

closed 2020-02-27 by carlescufi