zephyrproject-rtos/zephyr #10832: k_sleep should return information that thread was woken up

pdunaj, 2018-10-25:
At the moment `k_sleep` returns void so it is not possible to determine if thread waited for expected time or was woken up by `k_wakeup`.

It seems that at the moment there is no easy way to obtain this as current timeout is removed from the list on wakeup and information is lost. Additional variable could be added to the structure but it is also possible to pass this information in `t->dticks`: set it to new `_WAKEUP` value instead of `_INACTIVE` which would mean the same but pass information that `k_sleep` was aborted.

pdunaj, 2018-10-25:
@pizi-nordic

pizi-nordic, 2018-10-25:
Does posix-like behaviour (return number of ms we still have to wait in order to reach timeout) satisfies your needs?

pdunaj, 2018-10-25:
@pizi-nordic , sure :)