zephyrproject-rtos/zephyr #10845: ext: hal: nxp: Use ARRAY_SIZE helper macro

himanshujha199640, 2018-10-25:
CC @MaureenHelm 
```
himanshu@himanshu-Vostro-3559:~/zephyr$ ./scripts/coccicheck --cocci=scripts/coccinelle/array_size.cocci --mode=report --verbose=1 ext/hal/nxp/

Please check for false positives in the output before submitting a patch.
When using "patch" mode, carefully review the patch before submitting it.

Processing array_size.cocci
with option(s) " --no-includes --include-headers"

Message example to submit a patch:
 Use ARRAY_SIZE instead of dividing sizeof array with sizeof an element

 The semantic patch that makes this report is available
 in scripts/coccinelle/array_size.cocci.

 More information about semantic patching is available at
 http://coccinelle.lip6.fr/

Semantic patch information:
 This makes an effort to find cases where ARRAY_SIZE can be used such as
 where there is a division of sizeof the array by the sizeof its first
 element or by any indexed element or the element type. It replaces the
 division of the two sizeofs by ARRAY_SIZE.

Running (4 in parallel): /usr/local/bin/spatch -D report --no-show-diff --very-quiet --cocci-file scripts/coccinelle/array_size.cocci --no-includes --include-headers --patch /home/himanshu/zephyr --dir ext/hal/nxp/ --jobs 4 --chunksize 1
ext/hal/nxp/mcux/drivers/kinetis/fsl_uart.c:158:49-50: WARNING: Use ARRAY_SIZE
ext/hal/nxp/mcux/drivers/kinetis/fsl_ftm.c:86:47-48: WARNING: Use ARRAY_SIZE
ext/hal/nxp/mcux/drivers/lpc/fsl_wwdt.c:66:49-50: WARNING: Use ARRAY_SIZE
ext/hal/nxp/mcux/drivers/imx/fsl_adc_etc.c:43:53-54: WARNING: Use ARRAY_SIZE
ext/hal/nxp/mcux/drivers/kinetis/fsl_ltc.c:3004:45-46: WARNING: Use ARRAY_SIZE
ext/hal/nxp/mcux/drivers/kinetis/fsl_i2c.c:208:49-50: WARNING: Use ARRAY_SIZE
ext/hal/nxp/mcux/drivers/lpc/fsl_sctimer.c:86:47-48: WARNING: Use ARRAY_SIZE
ext/hal/nxp/mcux/drivers/lpc/fsl_mrt.c:65:47-48: WARNING: Use ARRAY_SIZE
ext/hal/nxp/mcux/drivers/imx/fsl_dcp.c:928:45-46: WARNING: Use ARRAY_SIZE
ext/hal/nxp/mcux/drivers/kinetis/fsl_tpm.c:67:47-48: WARNING: Use ARRAY_SIZE
ext/hal/nxp/imx/drivers/i2c_imx.c:116:56-57: WARNING: Use ARRAY_SIZE
ext/hal/nxp/imx/drivers/i2c_imx.c:119:41-42: WARNING: Use ARRAY_SIZE
ext/hal/nxp/mcux/middleware/wireless/framework_5.3.3/XCVR/MKW41Z4/fsl_xcvr.c:342:59-60: WARNING: Use ARRAY_SIZE
ext/hal/nxp/mcux/drivers/lpc/fsl_ctimer.c:74:53-54: WARNING: Use ARRAY_SIZE
```

pfalcon, 2018-10-29:
@himanshujha199640: You understand that everything in ext/ comes from 3rd-party sources, right? So, it's outside of our control and these reports are effectively "false positives", as the output above says.

galak, 2018-10-29:
> @himanshujha199640: You understand that everything in ext/ comes from 3rd-party sources, right? So, it's outside of our control and these reports are effectively "false positives", as the output above says.

I suggested that @himanshujha199640 open issues to report what he was seeing so at least there was visibility to those that might care.

pfalcon, 2018-10-29:
@himanshujha199640 

> I will now stop sending changes for ext/ as it seems uninteresting to the community.

Well, if you had an arrangement with @galak regarding doing that, which has now been explained, then please continue. My note was not to discourage you, but give a warnings that there may be a long path from finding some possible things to improve to them actually being improved.

As for "community", this specific patchset targets internal HAL maintained by NXP. Only NXP representatives may say whether it's useful to them. I personally agree with your analysis and will be thrilled if @MaureenHelm or other NXPers would confirm that the next version of MCUX will come with the changes suggested.

So again, sorry for the false alert, wasn't intended as such.

himanshujha199640, 2018-10-30:
> @himanshujha199640
>
>
> Well, if you had an arrangement with @galak regarding doing that, which has now been explained, then please continue. My note was not to discourage you, but give a warnings that [...]

I'm not discouraged! 
It's just that I don't want to waste either of our time pushing issues
which have a low priority.

> As for "community", this specific patchset targets internal HAL maintained by NXP. Only NXP representatives may say whether it's useful to them. I personally agree with your analysis and will be [...]

OK.
As @galak pointed to the 3rd party repo, you can also point me to it, and I will push
at that repo instead.

> So again, sorry for the false alert, wasn't intended as such.

I work with the linux kernel community, so used to it ;)
https://lore.kernel.org/lkml/20170829142011.76w67w64a7lheef4@piout.net/

nashif, 2019-02-19:
not a zephyr bug.

closed 2019-02-19 by nashif