zephyrproject-rtos/zephyr #10857: Migrating from CMSIS-Core to DeviceTree results in loss of type information

SebastianBoe, 2018-10-25:
We are in the early process of migrating from retrieving peripheral base addresses from CMSIS-Core to retrieving addresses from DeviceTree.

This has an unfortunate side-effect, namely, CMSIS drivers don't actually usually reference the u32_t base address itself. But a struct typed address that has register names and offsets encoded in it.

This results in DT drivers writing code like this:

`NRF_UART_Type * uart0_addr = (NRF_UART_Type *)CONFIG_UART_0_BASE;`
https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/serial/uart_nrfx_uart.c#L16

whereas with CMSIS one could omit this, and directly reference a symbol with type informaiton,
like NRF_UART0 for instance.

The approach taken in uart_nrfx_uart.c doesn't scale. We need to find out how we can replace the CMSIS base addresses with DT without having to write code like the above in all drivers.

SebastianBoe, 2018-10-25:
FYI: @anangl @barsok

galak, 2018-10-26:
I'm not sure I understand what feature from CMSIS you want?  Is it having global defines like NRF_UART0, NRF_UART1, etc..? that are a pointer to a struct for the registers of that device?

SebastianBoe, 2018-10-26:
> Is it having global defines like NRF_UART0, NRF_UART1, etc..? that are a pointer to a struct for the registers of that device?

Yes.

galak, 2018-10-26:
> Yes.

Why do we need such a thing?

SebastianBoe, 2018-10-26:
Because CMSIS drivers don't actually usually reference the u32_t base address itself. But a struct typed address that has register names and offsets encoded in it.

Not having access to this symbol results in DT drivers writing code like this:

NRF_UART_Type * uart0_addr = (NRF_UART_Type *)CONFIG_UART_0_BASE;
https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/serial/uart_nrfx_uart.c#L16

whereas with CMSIS one could omit this, and directly reference a symbol with type information,
like NRF_UART0 for instance.

The approach taken in uart_nrfx_uart.c doesn't scale. We need to find out how we can avoiid writing code like the above in all drivers.

SebastianBoe, 2018-10-29:
Code readability.

Since we no longer have typed base addresses, every reference to
hardware registers is going to look slightly different. E.g. in
one source file we might be calling it 'aurt0_addr', in another
'UART0_ADDR', in a third 'UART0', in a fourth, 'NRF_UART0_ADDR'
and so on.

With CMSIS you only needed to know whether a vendor distributed
CMSIS files with it's vendor prefix, or without, and would be
able to predict what the symbol would be called for any
peripheral from that vendor.

This was possible because the typed addresses followed the
pattern:

[<DeviceAbbreviation>_]<peripheral><instance_number>

E.g. for nordic it is

NRF_TIMER0

and for STM it is

TIM0

(STM has opted out of using the DeviceAbbreviation)

But with Device Tree there is no support and no pattern, each
source file is left to invent it's own scheme.

Note that existing ARM HAL's use the base addresses from CMSIS
instead of from DeviceTree. The CMSIS itself can be thought of as
the thinnest and lowest level of this HAL. And it happens to be
that all the arguments that Maureen and Frank made in favour of
HALs apply also to the typed base address from CMSIS.

"
For Zephyr, SoC vendor HALs offer several advantages, including:

Code is maintained elsewhere, meaning that there is less to
write, review, and maintain. What’s more, permissive licensing
and regular vendor-provided updates reduce overhead.

Code is used elsewhere, bringing a higher level of maturity, as
well as more extensive QA testing.

Expanded levels of abstraction provided by HALs allow for a range
of new or reusable Zephyr code.  These advantages, combined with
the expediency offered by SoC-provided HALs, make integrating
HAL-driven abstraction a logical choice for the Zephyr Project.
"
-- https://www.zephyrproject.org/zephyr-leverages-hals-to-accelerate-viability/

I can't see any solution to this problem apart from the
straightforward solution of allowing the use of typed addresses
from CMSIS, just as we are allowing the use of higher level HAL's
that eventually uses CMSIS, and as we are allowing the use of the
types themselves.

galak, 2018-10-29:
But should drivers be encapsulating the access to the HW registers?  As such we're going to end up with code generating what the driver desires the name to be called.

I'm also never been a fine of having the same define mean 2 different things based on the SoC.  We also have to think across arches w/regards to Zephyr.

I guess I'm still trying to understand what functionality you are wanting to expose to the developer and which developer?  An application developer or someone writing a driver for Zephyr

SebastianBoe, 2018-10-30:
> But should drivers be encapsulating the access to the HW registers? As such we're going to end up with code generating what the driver desires the name to be called.

Sorry, I don't understand what this means.

> I'm also never been a fine of having the same define mean 2 different things based on the SoC. We also have to think across arches w/regards to Zephyr.

I'm not sure what this means either. Which define are you talking about? Note that the base addresses have become more consistent across platforms with the DeviceTree migration. But drivers don't traditionally use the base addresses, and the typed addresses, which are used, have become less consistent in Zephyr from the DeviceTree migration, so as a whole things have becomes less consistent.

I would like to ask if you understand the arguments made for allowing referencing CMSIS typed addresses, and if so if we can allow this? If there was any confusion in what I wrote I can try to elaborate.

SebastianBoe, 2018-10-30:
> I guess I'm still trying to understand what functionality you are wanting to expose to the developer and which developer? An application developer or someone writing a driver for Zephyr

Any developer that needs to access peripheral registers. So both application developers and driver developers. Peripheral registers are accessed by application developers when, for whatever reason,
the driver does not support the application's use-case, and the application developer needs to access HW directly (through a typed address).

SebastianBoe, 2018-10-30:
> what functionality you are wanting to expose to the developer and which developer? 

In embedded, firmware does memory mapped IO to HW peripheral registers (addresses). I want the ability to reference these registers in a human-readable way.

E.g.

NRF_POWER->SYSTEMRESET = 1; // Do a soft reset of the system

pabigot, 2018-11-19:
The Nordic headers that provide the CMSIS structure declarations have things like:

```
#define NRF_UART0 ((NRF_UART_Type*) NRF_UART0_BASE)
#define NRF_POWER ((NRF_POWER_Type*) NRF_POWER_BASE)
```

which seems functionally the same as:

```
static NRF_UART_Type *const uart0_addr = (NRF_UART_Type *)DT_UART_0_BASE;
static NRF_POWER_Type *const power = (NRF_POWER_Type *)DT_POWER_BASE;
```

I too absolutely want to use the CMSIS structures to directly manipulate the peripheral registers, but I don't see how getting the base address of the peripheral instance to use from device tree instead of the vendor's header makes that difficult.  (Except in cases like the CLOCK/POWER module which isn't currently identified in the Nordic device trees; but that can be fixed.)

mbolivar-nordic, 2020-10-16:
Can this be closed? I don't think this is "fixable" and I think it hasn't proven to be a problem in practice.

erwango, 2020-12-16:
@mbolivar-nordic agree to close. @SebastianBoe feel free to re-open.

closed 2020-12-16 by erwango