zephyrproject-rtos/zephyr #10878: errno for error codes for key management break on newlib

galak, 2018-10-26:
The following added errno values aren't defined by newlib and thus cause issues when building with newlib:

```
/* specific encryption errno values */
#define ENOKEY 126       /* Required key not available */
#define EKEYEXPIRED 127  /* Required key not available */
#define EKEYREVOKED 128  /* Key has been revoked */
#define EKEYREJECTED 129 /* Key was rejected by service */
```

galak, 2018-10-26:
@nilsMontenegro can you look at this.

nilsMontenegro, 2018-10-26:
Yes. Can we define them in newlib as well? Would that be a pull request to the zephyr-SDK?

galak, 2018-10-26:
> Yes. Can we define them in newlib as well? Would that be a pull request to the zephyr-SDK?

defining them in newlib isn't an option.  We'd need a zephyr specific header for such extensions if we want to keep these defines.

galak, 2018-10-29:
This has been partially addressed by PR #10881, and the rest with PR #10879

nilsMontenegro, 2018-10-29:
@galak Yes, with these PRs in, that will not be a problem any more.

PR #10879 (2018-10-29): Remove errno defines not used (introduced for bluetooth l2cap)
The error codes for key management where put into errno.h, however if we
build with newlib these defines are visibile (since we get errno.h from
newlib's headers).  Move these defines to l2cap as its the user of them.

Fixes: #10878
closed 2018-10-29 by carlescufi