zephyrproject-rtos/zephyr #10884: Make functions related to context switching add implicit barriers

pdunaj, 2018-10-26:
To make sure that code is not reordered around functions that are critical for correct behavior of multi-threading, such functions should add barriers (compiler one or in case of MP systems proper hw instruction).

This is mostly important for locks of all kinds. (E.g. Linux kernel documentation states that locks add barriers).
This is not however the only case, consider following code:
```
while (checkme) {
   k_sleep(FEW_MS);
}
```
In normal case a function call to `k_sleep` will ensure that value of `checkme` is reloaded (as this is a call to external linkage function). The problem will appear when lto is enabled. In this case k_sleep code can be imported and reload of `checkme` optimized out.
Zephyr should ensure that (if not all) then some of the kernel functions will be implicit barriers. This should also be documented.

carlescufi, 2018-10-27:
Note that we do have this in irq_lock/unlock at least for Arm

pdunaj, 2018-10-29:
@carlescufi , I just checked and you are right.
Maybe we should add information in the documentation about functions with implicit barrier.
I read comment in `k_sleep` code explaining the volatile used there and I think such item is needed.

pizi-nordic, 2018-10-29:
@pdunaj is correct that not all cases are covered. For example on SMP, the irq_lock() is a call to global _smp_global_lock() and there is no compiler barrier on the call level. POSIX also might have some trouble caused by ordering.