zephyrproject-rtos/zephyr #10894: Add dts support to SPI slave drivers (CC1200_SPI), (CC2520_SPI) & tests (tests/drivers/spi/spi_loopback)

galak, 2018-10-29:
Issue #10636 exposed a problem that a few cases that SPI slave drivers aren't converted to DTS.  Doing a grep in the tree for SPI_._NAME or SPI_DRV_NAME we come up with:

```
boards/x86/quark_se_c1000_devboard/Kconfig.defconfig:	default SPI_1_NAME
boards/x86/quark_se_c1000_devboard/Kconfig.defconfig:	default SPI_1_NAME
drivers/ieee802154/Kconfig.mcr20a:	default SPI_0_NAME
tests/drivers/spi/spi_loopback/Kconfig:	default SPI_1_NAME
drivers/wifi/winc1500/Kconfig.winc1500:config WIFI_WINC1500_SPI_DRV_NAME
```

The mcr20a is fine as its in a `if !HAS_DTS_SPI` block, the quark_se_c1000 are for the CC1200_SPI & CC2520_SPI drivers.

We need to convert these slave drivers to use DTS.

galak, 2018-10-29:
@vanwinkeljan can you look at adding DTS support to drivers/display/display_ili9340.c.  Thanks.

galak, 2018-10-29:
@mbolivar can you add dts support to WS2812 & LPD880x

galak, 2018-10-29:
@tbursztyka can you look at converting adc_ti_adc108s102 to device tree.

galak, 2018-11-06:
> @mbolivar can you add dts support to WS2812 & LPD880x

Any update?  This drivers will soon not be usable once all SPI controllers use DTS.

galak, 2018-11-06:
@agansari can you look at converting the eth_enc28j60 driver over to use DTS.

galak, 2018-11-06:
PR #11165 to convert over BMI160 Sensor

carlescufi, 2018-12-05:
@anangl  have you taken a look at `drivers/bluetooth/hci/spi.c` ?

galak, 2019-01-09:
> @anangl have you taken a look at `drivers/bluetooth/hci/spi.c` ?

@anangl / @carlescufi any update?

carlescufi, 2019-01-10:
@galak not from me, no. I hope that @anangl can update us.

anangl, 2019-01-11:
@galak
 
> > @anangl have you taken a look at `drivers/bluetooth/hci/spi.c` ?
> 
> @anangl / @carlescufi any update?

I'm working on it. So far I found only one thing that is missing on the master side (in `drivers/bluetooth/hci/spi.c`). The CS line [is not asserted](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/bluetooth/hci/spi.c#L131) in the general case, there is only some specific steering of it for BlueNRG. But it's not that trivial to get this information from DTS since this line may be (at least for Nordic SoCs it is) configured in `cs-gpios` property in the parent node. Configuration of other settings is already in place (through [the "zephyr,bt-hci-spi" node](https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/96b_carbon/96b_carbon.dts#L89)).
On the slave side (in `samples/bluetooth/hci_spi`) to get [the relevant settings](https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/bluetooth/hci_spi/src/main.c#L51) I will follow the way you started in https://github.com/galak/zephyr/commit/f30bc9ea7033a1b63d5dde87ae765f2130050128, but I think that instead of adding the `spi-slave` property and setting `#address-cells` to `0` I will use a separate binding for "nordic,nrf-spis" node that will inherit from "spi-device". But first I need to make the hci_spi example working. Currently it is not handling correctly SPI transactions as it does not configure SPI to work in the slave mode, so the whole communication does not start. But even after I corrected this I still get some "invalid message type" errors. I need to debug it further.

galak, 2019-01-15:
Converted BME280 in PR #12496

galak, 2019-03-08:
PR #14187 handles the last reference, otherwise all controllers and spi clients are using DTS so closing this out.

closed 2019-03-08 by galak