zephyrproject-rtos/zephyr #10898: Add Nordic GPREGRET retrieval/setting functions

thedjnK, 2018-10-29:
This is a request to add support for the Nordic general purpose retention registers so that they can be read or set from Zephyr, so that a Zephyr-based application can set the state and reboot and detect what state it was in before resetting, it also allows for communication with other applications on the module which are not Zephyr based i.e. a bootloader

SebastianBoe, 2018-10-29:
For the record, be aware that it is possible to access these registers through "CMSIS" aka the "MDK" or nrf.h.

Something like

#include "nrf.h"

NRF_POWER->GPREGRET[0]

carlescufi, 2018-10-30:
@thedjnK Please see this pull request:

https://github.com/zephyrproject-rtos/zephyr/pull/10900

This fixes your request through the use of Zephyr's reset type.

closed 2018-10-30 by carlescufi