zephyrproject-rtos/zephyr #10917: Convert iwdg_stm32 SAM Watchdog driver to new API

galak, 2018-10-29:
We have a new driver API for watchdog, the old API has been deprecated for several releases now. We need to convert the driver to use the API or remove it.

galak, 2018-10-29:
@apodogrocki can you look at update the driver?

qianfan-Zhao, 2018-11-14:
I will do it

apodogrocki, 2018-11-14:
@qianfan-Zhao I've already started working on that. If you would like you can think of adding _window mode_ to the Independent Watchdog driver. It has never been activated, but can be useful. That probably requires separate PR.

@erwango @galak There is also a question how to provide support for System Window Watchdog. All watchdog devices are named "WATCHDOG_0" at the moment. In case of STM32 can both IWDG & WWDG co-exist in the same time? Or Kconfig.stm32 should contain _choice menu_ to select desired watchdog type?

qianfan-Zhao, 2018-11-14:
> here is also a question how to provide support for System Window Watchdog. All watchdog devices are named "WATCHDOG_0" at the moment.

Currently the name of watchdog are defined in Kconfig. Based on the comments in #11331 we should remove the Kconfig options and make DT define the name of Kconfig.

Others SoC such as NRF has this lines in dts_fixup.h:
```
#define CONFIG_WDT_0_NAME		DT_NORDIC_NRF_WATCHDOG_40010000_LABEL
```

If there has 2 watchdogs, such as IWDG and WWDG, i think it OK to define the difference name in DT:
```
#define CONFIG_WDT_0_NAME DT_STM32_IWDG_XXX_LABEL
#define CONFIG_WDT_1_NAME DT_STM32_WWDG_XXX_LABEL
```

apodogrocki, 2018-11-14:
@qianfan-Zhao Ok, I see that you have better insight in the current Zephyr code than me. So maybe it is correct that you should take care of this task. After all it seems that @galak has already made a decision.

galak, 2018-11-14:
> @qianfan-Zhao Ok, I see that you have better insight in the current Zephyr code than me. So maybe it is correct that you should take care of this task. After all it seems that @galak has already made a decision.

Sorry, I got confused by the thread, and so I assigned it to @qianfan-Zhao.  I'm happy for either of you guys to get this driver updated.  I don't think @apodogrocki was a collaborator of the project, so couldn't add you :).

commit 107375da (2019-02-02): drivers: iwdg_stm32: Convent STM32 watchdong driver to new API

Fixes: #10917
closed 2019-02-02 by nashif