zephyrproject-rtos/zephyr #10930: Extending string formatting function

nordic-krch, 2018-10-30:
Today we use 2 different string formatting functions (at least in logger and shell):
- _prf from minimal libc if newlibc is not used
- _vprintk otherwise

It has been started by @jhedberg in #9090. Then briefly touched with @jhedberg, @mike-scott, @pizi-nordic and @jarz-nordic to extend string formatting function to support zephyr specific format specifiers for things like BLE or NET addresses (like linux do https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/core-api/printk-formats.rst#n199) . Ideally, we should extend that only in one implementation since i expect that number of custom specifiers will grow and support both will cause issues. Then question is: which implementation should we support? Should we extract string formatting from minimal libc and make it common for printk, logger, shell and minimal libc?

Another thing is extending API of string formatting function. Currently we have:
`int _prf(int (*func)(), void *dest, char *format, va_list vargs);`
but for logger case (processing messages with known number of arguments) I would like to see:
`int _prf(int (*func)(), void *dest, char *format, u32_t *args, u32_t nargs);`
Currently we have huge switch-case in `log_output.c` that has to be extended every time we extend number of arguments supported by the logger. However, it is actually worse, I've found out that code generated for this switch case is huge:
- 1,7k for cortex m4 (after some tweaks reduced to 0,8k)
- 2,2k for quark_se_c1000_devboard (after tweaks reduced to 1,1k)

nordic-krch, 2018-10-30:
@nashif @carlescufi could you forward this. I'm not sure who was responsible for string formatting functions, printk, minimal libc etc.

nashif, 2021-06-16:
outdated

closed 2021-06-16 by nashif