zephyrproject-rtos/zephyr #10937: boilerplate.cmake and environment variable

FlorianFouillet, 2018-10-30:
hi everyone, I am trying to use the zephyr stack and I am facing the same issues for the last days..

I am following the Linux development guide and doing every steps. then when I am trying to build the hello world example I keep having the cmake issue:

```
$ sudo cmake -DBOARD=atsamd21_xpro ../.
CMake Error at CMakeLists.txt:3 (include):
  include could not find load file:
    /cmake/app/boilerplate.cmake

```

I have the file boilerplate but I think the issue is with the environment variable. 

CmakeLists for hello_world:

```
cmake_minimum_required(VERSION 3.8.2)
include(**$ENV{ZEPHYR_BASE}**/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)
project(hello_world)
target_sources(app PRIVATE src/main.c)
```

I am sourcing the zephyr-env.sh and I also tried by exporting the variable myself:

`export ZEPHYR_BASE=/home/user/zephyr`

but I keep having the same issue.. I tried with my current dev VM (ubuntu 17.10) I then tried on another VM (ubuntu 16.04) and I just tried again with a fresh VM (ubuntu 16.04) but I keep facing the same issue.

I have the correct version of cmake (3.8.2), I followed every steps of the dev guide.

Thank you for your help!

SebastianBoe, 2018-10-30:
Perhaps the env vars are different when sudo is used. Try without sudo.

carlescufi, 2018-10-30:
@FlorianFouillet yes, as @SebastianBoe says, you really should not be building with `sudo` in the first place unless you have a very good reason for it.

closed 2018-10-30 by FlorianFouillet

carlescufi, 2018-10-30:
@FlorianFouillet Glad to hear this worked for you. Remember to install dependencies with `pip install --user` so that they are placed in the user folder and not a global one.