zephyrproject-rtos/zephyr #10942: Declaring an API Stable

nashif, 2018-10-30:
from the API meeting in Novembe 2017 this was the definition of stable APIs:

- When introduced: unstable by default
- At least 2(?) implementations of the API for different HW ->  generic. 
- (Multiple) Applications in the case of HW agnostic APIs
- Dedicated model still valid, declare API for HW features (no catch-all API functions for HW specific features)
- Stable: basic requirements for an API + at least one (?) release a generic

We need to formalise this and make it as part of the documentation.

galak, 2018-10-30:
Test of the full API should be part of the requirements for 'stable' classification.

ceolin, 2018-10-30:
Question related with this matter. How is our policy about API breaks ?
Asking because With MISRA-C changes we'll need break some API due things like
identifier clashes.

pfalcon, 2018-11-06:
@ceolin:

> How is our policy about API breaks ?

Well, we break APIs all the time. The whole "LTS and stable API" matter is about stopping doing that routinely (apparently, by having APIs which don't suck). When we do that, we try to deprecate old API elements (functions, structures, etc.), while introducing new. Well, we **do** that for public application APIs. For internal APIs, like drivers API, doing that (much) harder, so exceptions are more frequent.

Just right now you can find at least couple of PRs to deprecate something, and recently there was a series of PRs to garbage-collect stuff deprecated in older releases.

> Asking because With MISRA-C changes we'll need break some API due things like
identifier clashes.

So we should:

1. Do that ASAP.
2. Do that consistently, pervasively, and in a way to really find the best choice and make it stable, not break again.

Of course, p.1 and p.2 conflict on the surface, but "sooner and right" is always better than "later and in whatever way".

pfalcon, 2018-11-06:
> When introduced: unstable by default

Perhaps worth being more specific: any new API should start as EXPERIMENTAL in Kconfig. Then likely, should stay that way for N releases. (Where 1 is over-optimistic, at least for 3-mon releases we used to have).

> At least 2(?) implementations of the API for different HW -> generic.

2 implementations for stable API? LOL

> (Multiple) Applications in the case of HW agnostic APIs

Good mention. For example, for POSIX FS, there're tests, but zero samples.

> Dedicated model still valid, declare API for HW features (no catch-all API functions for HW specific features)

This point needs clarifications.

> Stable: basic requirements for an API + at least one (?) release a generic

Again, needs clarification. I guess this touches on what I mention above: how much API spends in each state. But there should be more states: EXPERIMENTAL -> "non-stable" -> stable.

galak, 2018-11-06:
Here are a few classes to start with: stable, experimental, deprecated, etc.

pfalcon, 2018-11-07:
> from the API meeting in November 2017

Btw, is this correct? I.e. this captures the status of the discussion of the matter done a year ago, and asks if it's good enough to be captured in the docs by now?

(It would be easy to think that it's a typo, but then I know that the API meeting was yesterday, Nov 6, they're weekly, so the previous one wasn't even in November).

pfalcon, 2018-11-07:
I can also propose a "case study" based on APIs I'm familiar with:

1. UART driver API has 27 implementations. Is it stable? Well, there're 2 sub-APIs: synchronous I/O and interrupt-driven async I/O. The first is definitely stable and adheres to all requirements (like, being easy to implement for board bring-up). The interrupt-driven async I/O API isn't stable at all. It's not documented thoroughly enough. There're now reports that "subsystems" written on top of it have issues with particular platforms (https://github.com/zephyrproject-rtos/zephyr/issues/10726), which may be due to bug in that platform's driver, or may be because API is on the wrong level which caters for issues like that. There're criticism that it's too low-level indeed (https://github.com/zephyrproject-rtos/zephyr/pull/10820). There're issues with error handling (https://github.com/zephyrproject-rtos/zephyr/pull/11039), which can be seen as hw-specific issue again, but actually API doesn't specify requirements/API contract of handling those.

So, at 27 implementations, it's definitely at p.2 definition above, but far from p.1.

2. Sensor API. 48 implementations. Some time ago I looked at that API, found some things more or less surprising (negative values representation is what I remember for example). Those were raised for discussion, and at least were documented explicitly. Now I heard @MaureenHelm questions again whether decimal-based representation of values is adequate. So, with so many implementations, it's not exactly criticism-free.

nashif, 2018-11-07:
> Even for the last "spur of the day" usage, having just 2 implementations doesn't seem like serious criteria.

**At least 2(?) implementations of the API for different HW -> generic.**

So lets analyse this:

- at least = this is fixed to X implementations, this will depend on the domain and subsystem. So being minutes from a live discussion "at least" here signifies that we know that some domains will need more.

- 2(?): Meaning that the number was put there as a placeholder open for discussion and not finalised.

nashif, 2018-11-07:
> Btw, is this correct? I.e. this captures the status of the discussion of the matter done a year ago, and asks if it's good enough to be captured in the docs by now?

yes, this is correct, pulled those bullets from first API meeting we had 1 year ago. They were in raw firm and copy/pasted as is and to pick up and continue the discussion. Those are discussion minutes, **not** decisions.

dbkinder, 2019-02-27:
Does https://docs.zephyrproject.org/latest/development_process/api_lifecycle.html address this issue?

dbkinder, 2019-06-18:
@nashif @carlescufi What needs to be done here to address this issue?

carlescufi, 2019-06-18:
This is already done actually, because it was cleanly documented in the page that @dbkinder linked: https://docs.zephyrproject.org/latest/development_process/api_lifecycle.html

closed 2019-06-18 by carlescufi

dbkinder, 2019-06-18:
@carlescufi thanks!