zephyrproject-rtos/zephyr #10943: Objects in a custom RAM section defined in custom-sections.ld don't get initialized

michiwend, 2018-10-30:
Working on an ARM cortex_m platform (NRF52840), when adding objects to a custom RAM section defined in `custom-sections.ld` in my project folder the objects don't get initialized to their initial values. However when I add the same section definition to `include/linker/common-ram.ld` it works as expected. I'm not sure whether I'm doing it wrong, it's not what the `custom-sections.ld` is for or if it is an actual bug?

I define my custom section as follows:
```
SECTION_DATA_PROLOGUE(_example_area, (OPTIONAL), SUBALIGN(4))
{
  _example_list_start = .;
  KEEP(*(SORT_BY_NAME("._example.static.*")))
  _example_list_end = .;
} GROUP_DATA_LINK_IN(RAMABLE_REGION, ROMABLE_REGION)
```

The following code is a complete example to test the custom section:
```c
#include <zephyr.h>
#include <misc/printk.h>

#define EXAMPLE_DEFINE(_name, _value)			\
	struct example _name				\
	__in_section(_example, static, _name) = {	\
		.value = _value, 			\
	}						\

struct example {
	uint32_t value;
};

extern struct example _example_list_start[0];
extern struct example _example_list_end[0];

static inline struct example *example_get(u32_t idx) {
	return &_example_list_start[idx];
}

static inline int example_count_get(void) {
	return ((void *)_example_list_end - (void *)_example_list_start) /
		sizeof(struct example);
}

EXAMPLE_DEFINE(foo, 0xdeadbeef);

void main(void)
{
	printk("number of objects: %d\n", example_count_get());
	printk("value of foo: %x\n", foo.value);
}

```

Output of the example application when section is defined in `common-ram.ld`:
```
number of objects: 1
value of foo: deadbeef
```

Output of the example application when section is defined in `custom-sections.ld`:
```
number of objects: 1
value of foo: 0
```
*Note:* When first running the same application with the section being defined in `common-ram.ld` and then switching over to `custom-sections-ld`, foo.value will still be whatever it was when the section was defined in `common-ram.ld` rather then 0.

SebastianBoe, 2018-10-30:
"it's not what the custom-sections.ld is for or if it is an
actual bug?"

I wouldn't say it's a bug. For that section to be copied from
flash to ram it has to be placed in a location that the kernel
will run memcpy on (see _data_copy() in init.c).

I think you are probably looking for CUSTOM_RWDATA_LD.

```
#ifdef CONFIG_CUSTOM_RWDATA_LD
/* Located in project source directory */
#include <custom-rwdata.ld>
#endif
```

as that will be placed somewhere that is copied from flash to
ram.

I see there is also

```
#ifdef CONFIG_APPLICATION_MEMORY
	(void)memcpy(&__app_data_ram_start, &__app_data_rom_start,
		 ((u32_t) &__app_data_ram_end - (u32_t) &__app_data_ram_start));
#endif
```

Not familiar with either one, so I can't tell you which one you
would want to use.

closed 2018-10-31 by SebastianBoe

michiwend, 2018-10-31:
@SebastianBoe thanks for your help. Putting the definition inside of the custom-rwdata.ld works. But since it is already included into a section i just kept the inner three lines like so:
```
_example_list_start = .;
KEEP(*(SORT_BY_NAME("._example.static.*")))
_example_list_end = .;
```

HenrikKaa, 2024-06-07:
I ran into this same problem and just cannot solve it. Zephyr has changed so much since about 1.1.0 that this isn't any help anymore. A new example would be greatly appreciated.