zephyrproject-rtos/zephyr #10945: native_posix fails to compile

ochipara, 2018-10-30:
Hi,

I am trying to compile hello_world for native_posix on an os x (high sierra). I am able to use cmake to generate the build files, but it fails during compilation. I am getting the following error 

FAILED: zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj 
/usr/bin/gcc -DBUILD_VERSION=zephyr-v1.13.0-1408-g7700eb2a15 -DKERNEL -D_FORTIFY_SOURCE=2 -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -D__ZEPHYR__=1 -I../../../kernel/include -I../../../arch/posix/include -I../../../soc/posix/inf_clock -I../../../soc/posix/inf_clock/include -I../../../soc/posix/include -I../../../boards/posix/native_posix -I../../../include -I../../../include/drivers -Izephyr/include/generated -m32   -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /Users/ochipara/Working/embedded/r_and_r/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Werror=implicit-int -Wpointer-arith -ffunction-sections -fdata-sections -fno-freestanding -m32 -MMD -MP -include /Users/ochipara/Working/embedded/r_and_r/zephyr/arch/posix/include/posix_cheats.h -std=c11 -MD -MT zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj -MF zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj.d -o zephyr/CMakeFiles/offsets.dir/arch/posix/core/offsets/offsets.c.obj   -c /Users/ochipara/Working/embedded/r_and_r/zephyr/arch/posix/core/offsets/offsets.c
clang: error: unknown argument: '-fno-freestanding'

I remove the '-fno-freestanding' flag from ninja.build, then I am running into additional errors due to not finding the macros GEN_OFFSET_SYM and GEN_ABSOLUTE_SYM. Any help in getting the sample to compile for native_posix would be appreciated.

Thanks,
-- Octav

aescolar, 2018-10-31:
Hi @ochipara , we have tried to get native_posix working on mac OS and do know from a couple of people who tried that it does not work.

Emil-Gydesen-Bose, 2019-12-19:
@ochipara I see that the error is `clang: error: unknown argument: '-fno-freestanding'`

Have you tried compiling with GCC instead of clang? I don't own a mac myself, but I've heard that `gcc` symlinks to the clang (llvm) compiler by default on macs, so that you think you have GCC, but you don't. 

There's a couple of guides on how to download, build and install GCC on macs on the web, if you haven't already tried compiling with GCC.

marc-hb, 2019-12-19:
> @ochipara I see that the error is clang: error: unknown argument: '-fno-freestanding'

I suspect this particular error is  gone with a more recent XCode. On Mojave now:

```
$ clang --version
Apple LLVM version 10.0.1 (clang-1001.0.46.4)
$ clang --help | grep standing
  -ffreestanding          Assert that the compilation takes place in a freestanding environment
```

> I don't own a mac myself, but I've heard that gcc symlinks to the clang (llvm) compiler by default on macs, so that you think you have GCC, but you don't.

This is correct. Don't invoke `gcc` on a Mac, it's deceptive (unless you installed it yourself). Use `clang`.

> There's a couple of guides on how to download, build and install GCC on macs on the web, if you haven't already tried compiling with GCC.

The most likely easiest: https://formulae.brew.sh/formula/gcc

raveslave, 2019-12-20:
> I suspect this particular error is gone with a more recent XCode.   

I'm on latest os + xcode `Apple clang version 11.0.0 (clang-1100.0.33.16`, not getting the fno- error, but instead a bunch of `../kernel/include/kernel_offsets.h:27:1: error: unknown directive`

raveslave, 2020-01-18:
any plans to support LLVM/Xcode on macOS or should we use gcc for the native_posix use case (on mac)?

aescolar, 2020-01-20:
@raveslave No plans wrt macOS and native_posix

eanderlind, 2020-12-27:
Building `zephyr/samples/hello_world` with gcc (10.2 installed via homebrew, with a manual link `ln -s /usr/local/bin/gcc-10 /usr/local/bin/gcc`) on macOS 10.15.7 (ie only 64-bit supported) with 
`west build -b native_posix_64 -d build_posix64` 
selects correct gcc toolchain, however stumbles when compiling ``zephyr/arch/posix/core/offsets/offsets.c`with errors
```
../../../kernel/include/kernel_offsets.h:28:2: error: unknown directive
        .type   ___cpu_t_current_OFFSET,@object
        ^
../../../kernel/include/kernel_offsets.h:29:2: error: unknown directive
        .type   ___cpu_t_nested_OFFSET,@object
        ^
../../../kernel/include/kernel_offsets.h:30:2: error: unknown directive
        .type   ___cpu_t_irq_stack_OFFSET,@object
        ^    (truncated the rest..)
```
It looks like the struct member .type isn't defined for macOS. Any suggestions which include directive could be wrong and which .cmake file to modify? 
(-m64 flag was set, but zephyr/.config didn't have any flags containing `X86`)

Note: With same setup, `west build -b qemu_x86_64 -d build_qemu64` also fails - it incorrectly selects the 
gccarmemb toolchain even when creating a new build dir (and complains it doesn't understand `m64 flag).  
Logic for selecting target seems incorrect for this case.

Building `west build -b qemu_cortex_m3` is successful.

closed 2024-02-09 by aescolar