zephyrproject-rtos/zephyr #10952: Bluetooth Mesh: Disabling "CONFIG_BT_MESH_DEBUG_ACCESS" leads to wrong OpCode attached to sent messages

RonAmihai, 2018-10-31:
Me & my team are developing Zephyr Bluetooth mesh application.
Our application is working as expected in version (1.12). Currently we are using self provisioning in both server & client models if it matters.

After upgrading to Zephyr 1.13.0 - we've noticed that disabling the "CONFIG_MESH_DEBUG_ACCESS" causes our server model to attach WRONG opcode to sent messages using publication context.
(In our case, 0x00000000 and 0x00008813 repeatedly, instead of OpCode 0x00000040)

Enabling this debug configuration back "fixes" this issue.

Our client model receives the messages correctly regardless of his configuration - as long as the messages are sent with the correct opcode. So I'm pretty sure the issue is related to message publication.

I've attaches 2 examples of logs **of client model side** (server side logs without DEBUG_ACCESS is useless I guess...) to demonstrate the issue:

1) Situation when server side enables CONFIG_BT_MESH_DEBUG_ACCESS (Working state):
[Mesh Client (when server enables 'DEBUG_ACCESS').txt](https://github.com/zephyrproject-rtos/zephyr/files/2532408/Mesh.Client.when.server.enables.DEBUG_ACCESS.txt)

2) The other represents situation when the server disables CONFIG_BT_MESH_DEBUG_ACCESS configuration (Related issue):
[Mesh Client (when server disables 'DEBUG_ACCESS').txt](https://github.com/zephyrproject-rtos/zephyr/files/2532407/Mesh.Client.when.server.disables.DEBUG_ACCESS.txt)

jhedberg, 2018-11-08:
It appears that it's not just the opcode that's wrong, but a large part of the SDU is zeroes. Working:
```
[bt] [DBG] bt_mesh_model_recv: (0x20001b54) len 9: 4080454149e9431700
```
vs not working:
```
[bt] [DBG] bt_mesh_model_recv: (0x20001b54) len 9: 000000007187000000
```

Anyway, the receiving side gives only half of the picture. What about the sending side? How does the received SDU content compare with what the server-side logs claim to be sending?

FWIW, I've never heard or seen such an issue before, so I would have an initial bias at looking at the application code for bugs rather than the stack. It might also be some kind of stack overflow that causes corruption. That said, it could be something on the stack side as well.

Could you please also try with the very latest master branch?

RonAmihai, 2018-11-10:
Erasing the device's flash using nrfjprog utility (-e / --eraseall) actually solved this issue for us. 
Thanks for helping anyway.

closed 2018-11-10 by RonAmihai