zephyrproject-rtos/zephyr #10965: wncm14a2a modem should be moved to a shield configuration

nashif, 2018-10-31:
The modem is part of a shield, currently is hardcoded into the frdm_k64f board DTS. We should move it to a shield an make it available to other boards.

mike-scott, 2018-10-31:
@erwango What is your recommendation for establishing digital pin exports from the main board in DTS?

Under non-shield configuration:
```
       wncm14a2a {
               compatible = "wnc,m14a2a";
               label = "wncm14a2a";
               mdm-boot-mode-sel-gpios = <&gpioc 17 0>;
               mdm-power-gpios = <&gpiob 9 0>;
               mdm-keep-awake-gpios = <&gpioc 2 0>;
               mdm-reset-gpios = <&gpioc 12 0>;
               mdm-shld-trans-ena-gpios = <&gpioc 4 0>;
               mdm-send-ok-gpios = <&gpiod 0 0>;
               status = "ok";
       };
```

These should be exported in some way from the main board's DTS (note this doesn't work):
```
arduino_d10: <&gpiod 0 0>; {};
arduino_d9: <&gpioc 4 0>; {};
arduino_d8: <&gpioc 12 0>; {};
arduino_d6: <&gpioc 2 0>; {};
arduino_d2: <&gpiob 9 0>; {};
arduino_d1: <&gpioc 17 0>; {};
```

erwango, 2018-11-05:
@mike-scott 

There were on going work on dtc to enable similar stuff (https://elinux.org/images/d/d0/Panto.pdf) , but I don't know current status. I'll check on dtc status and come back to you.

There are some ways to get dtc compiled and have something like (in dts_compiled file). For instance:
```
arduino_spi: &spi1 {

	arduino_dx: pin {
		 gpios = <&gpioa 8 0>;
	 };
```
Would compile into
```
		arduino_spi: spi1: spi@40013000 {
			[...]
			arduino_dx: pin {
				gpios = <0x5 0x8 0x0>;
				phandle = <0x6>;
			};

			bt0: spbtle-rf@0 {
				[...]
				reset-gpios = <0x6>;
			};
		};
```

But then it requires some hack in extract_dts_include script to do the last step, would be better to have this done in dtc. So this would work, but would be a divergence from Linux. It seems we are a the edge of the dtc features here... Maybe @galak you have more inputs?
So either we chose our on way, waiting to have things integrated (maybe in a different way) in Linux, either we wait Linux. Let's check last dtc status before going further.

mike-scott, 2018-11-16:
Did we come to any resolution for making placeholders for the gpio pins:
Hack extract_dts_include script or make a change to dtc?

I didn't see @galak weigh in on either.

galak, 2018-11-16:
> Did we come to any resolution for making placeholders for the gpio pins:
> Hack extract_dts_include script or make a change to dtc?
> 
> I didn't see @galak weigh in on either.

what do you mean by place holders?

mike-scott, 2018-11-16:
I was explaining above, that along with the UART which can currently be configured in the shield dts as a placeholder like so:
```
&arduino_serial {
       wncm14a2a {
               compatible = "wnc,m14a2a";
               label = "wncm14a2a";
               status = "ok";
       };
};
```
This is "connected" to the K64F dts (exposes the fact that there's a serial on the arduino headers):
```
arduino_serial: &uart2 {};
```

There are also 5 GPIO pins which need to be handled by the modem driver.  These are not so easy to export in this way.  @erwango had a fairly good idea illustrated above.

Something like this connecting the Arduino D13 pin to the K64F's PTD1 pin:
```
	arduino_d13: pin {
		 gpios = <&gpiod 1 0>;
	 };
```

erwango, 2018-11-20:
@agross-linaro , did you had the chance to discuss this point at plumbers?

mike-scott, 2019-03-05:
RFC to discuss supporting GPIOs in a sane way for shields: https://github.com/zephyrproject-rtos/zephyr/pull/14057

mike-scott, 2019-05-22:
Closed by: https://github.com/zephyrproject-rtos/zephyr/commit/c0d9e7fcb5c4a3329273f8b137187abb288feda5#diff-a56763253b3f0eebe8c067f6a3ebba0b

closed 2019-05-22 by mike-scott