zephyrproject-rtos/zephyr #10967: FRDM-K64F and MCR-20A configuration does not work

finikorg, 2018-10-31:
Since some DTS stuff was changed I cannot use FRDM-K64F and MCR-20A

Errors are:
```
...
[00:00:09.123,543] <dbg> ieee802154_mcr20a.mcr20a_abort_sequence: CTRL1 0x39
[00:00:09.123,611] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.123,701] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.123,792] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.216,506] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.216,891] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.216,980] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.217,067] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.217,155] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.220,069] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.220,157] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.220,245] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.220,304] <dbg> ieee802154_mcr20a._irqsts1_event: Finished TxSeq
[00:00:09.220,309] <dbg> ieee802154_mcr20a.mcr20a_thread_main: WB: 0x0b | 0x00 | 0xf0
[00:00:09.220,380] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.220,468] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.220,556] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.333,849] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.333,937] <inf> spi_mcux_dspi.spi_context_cs_configure: CS control inhibited (no GPIO device)
[00:00:09.333,968] <err> ieee802154_mcr20a.mcr20a_tx: Timeout occurred, -11
[00:00:09.333,970] <err> wpanusb.tx: Error sending data, seq 44
...
```

finikorg, 2018-11-05:
@jfischer-phytec-iot have you tried this driver recently?

jfischer-phytec-iot, 2018-12-05:
I have looked at it and I found a few minor bugs in mcr20a driver but there is one ugly bug that I have been trying to find for a long time. It almost hits somewhere in spi_mcux_dspi and appears only in a configuration with mcr20a + ieee802154 + frdm_k64f. For example a have a build where spi driver drops first octet in the transfer (`spi_mcux_dspi.spi_mcux_transfer_next_packet: Transfer could not start`), same configuration works fine with a additional LOG_ERR in `spi_mcux_transfer_next_packet`.

jfischer-phytec-iot, 2018-12-06:
Update: depending on the build, the manifests as `spi_mcux_dspi.spi_mcux_transfer_next_packet: Transfer could not start`. I have debugged it for a while and with a breakpoints in `spi_mcux_master_transfer_callback`, the error will no longer occur. The bug also disappears if SPI clock is set to 1MHz or bus clock to 20MHz.

galak, 2019-03-26:
@jfischer-phytec-iot can you see if this is still an issue.

jfischer-phytec-iot, 2019-03-29:
tested with `samples/net/wpanusb` and `samples/net/sockets/echo_server`, can not reproduce the issue with current master, close it.

closed 2019-03-29 by jfischer-phytec-iot