zephyrproject-rtos/zephyr #10972: Modify socket samples to use user mode

laperie, 2018-10-31:
Since protected mode support was enabled in Zephyr for sockets (See https://github.com/zephyrproject-rtos/zephyr/issues/8203) at least some of socket-based samples need to be modified to support it.

pfalcon, 2018-11-07:
Ok, so as this was mentioned at yesterday's networking call, let's go forward with this, and let me record response to this ticket as a kind of walkthru.

1. Let's choose a random socket sample, e.g. samples/net/sockets/http_get , and build it for the default BOARD, which is qemu_x86.
2. Grepping outdir/qemu_x86/zephyr, there's CONFIG_USERSPACE=y. That fully corresponds to my expectations, as I "know" that qemu_x86 runs with userspace mode enabled for quite a while. That's why we have another, lesser known BOARD qemu_x86_nommu.
3. Let's how verify that "knowledge" and see where CONFIG_USERSPACE=y comes from:
~~~
zephyr/boards/x86/qemu_x86$ grep USERSPACE *
grep: doc: Is a directory
qemu_x86_defconfig:CONFIG_USERSPACE=y
qemu_x86_iamcu_defconfig:CONFIG_USERSPACE=y
~~~
4. So, the response to this ticket is that since https://github.com/zephyrproject-rtos/zephyr/pull/9179 was pushed, we built, and ran *all* socket samples in userspace mode, with API calls accessed using syscalls. (And yep, I don't know how other people, but I run them relatively regularly.)
5. For the avoidance of doubt, let's also grep the assembly:
~~~
        printf("Response:\n\n");                               
    4563:       68 bf de 01 00          push   $0x1debf        
    4568:       e8 43 d9 ff ff          call   1eb0 <iprintf>  
    456d:       5f                      pop    %edi
        __asm__ volatile ("mov %%cs, %[cs_val]" : [cs_val] "=r" (cs));
    456e:       8c c8                   mov    %cs,%eax
                            
K_SYSCALL_DECLARE6(K_SYSCALL_ZSOCK_RECVFROM, zsock_recvfrom, ssize_t, int, sock, void *, buf, size_t, max_len, int, flags, struct sock
    4570:       83 f8 2b                cmp    $0x2b,%eax
    4573:       75 26                   jne    459b <main+0x17c>
        __asm__ volatile("push %%ebp\n\t"
    4575:       31 db                   xor    %ebx,%ebx
    4577:       c7 45 f0 00 00 00 00    movl   $0x0,-0x10(%ebp)
    457e:       b9 ff 03 00 00          mov    $0x3ff,%ecx
    4583:       ba 60 03 40 00          mov    $0x400360,%edx
    4588:       be 92 00 00 00          mov    $0x92,%esi
    458d:       8b 45 e4                mov    -0x1c(%ebp),%eax
    4590:       89 df                   mov    %ebx,%edi
    4592:       55                      push   %ebp
    4593:       8b 6d f0                mov    -0x10(%ebp),%ebp
    4596:       cd 80                   int    $0x80
    4598:       5d                      pop    %ebp
~~~

So yeah, after that printf() we have call to recv(), which (after heavy macro machinery on the C side) is rendered as `int    $0x80`.

So, again, to summarize, whether to work in userspace mode or not, is selected by a particular machine, and for at least one, we've been working like that for quite a while. I know that there's work in parallel to enable userspace mode by default for ARM too.

Not closing right away, to avoid any confusion, because as I mentioned, my initial thought regarding this ticket was "that's how it works for quite a while". So, @laperie, please confirm that this addresses the topic.

pfalcon, 2018-11-12:
Hopefully it's answered.

closed 2018-11-12 by pfalcon