zephyrproject-rtos/zephyr #10990: ext: lib: crypto: mbedTLS: possible issue in platform.h glue

lemrey, 2018-11-01:
Line [226](https://github.com/zephyrproject-rtos/zephyr/blob/d70eb9853720bdae730fa598a87f72a96a302001/ext/lib/crypto/mbedtls/include/mbedtls/platform.h#L226) looks suspicious: the pattern has changed in the latest commit, but that's the only place where it changed. The patterns for printf, fprintf etc. remain unchanged.

The compiler emits a warning when compiling Zephyr's MCUBoot port:

`In file included from zephyr/ext/lib/crypto/mbedtls/library/oid.c:39:0:
zephyr/ext/lib/crypto/mbedtls/library/oid.c: In function ‘mbedtls_oid_get_numeric_string’:
zephyr/ext/lib/crypto/mbedtls/include/mbedtls/platform.h:226:28: warning: implicit declaration of function ‘MBEDTLS_PLATFORM_STD_SNPRINTF’ [-Wimplicit-function-declaration]

#define mbedtls_snprintf   MBEDTLS_PLATFORM_STD_SNPRINTF
                            ^
zephyr/ext/lib/crypto/mbedtls/library/oid.c:729:15: note: in expansion of macro ‘mbedtls_snprintf’
         ret = mbedtls_snprintf( p, n, "%d.%d", oid->p[0] / 40, oid->p[0] % 40 );`

SebastianBoe, 2019-03-01:
This warning breaks CI due to -Werror, looking at a fix ...

galak, 2019-03-07:
@SebastianBoe can we close this or still an issue?  Little confused about the MCUBoot reference.

SebastianBoe, 2019-03-07:
This can be closed. It was a bug that has now been fixed in MCUBoot.

closed 2019-03-07 by SebastianBoe