zephyrproject-rtos/zephyr #11007: net debug loggin: "log_strdup pool empty!"

pfalcon, 2018-11-01:
I've enabled CONFIG_NET_DEFAULT_LOG_LEVEL_DBG=y (with echo_server), and here's what I see:

~~~
[00:00:06.554,027] <dbg> net_ctx.net_context_bind: (0x20002510): Context 0x2000072c binding to UDP [::]:4242 iface 0x2000c5c0
[00:00:06.555,215] <dbg> net_conn.net_conn_register: (0x20002510): [3/10/17/0x05] remote 0x00000000/log_strdup pool empty!/0 
[00:00:06.556,034] <dbg> net_conn.net_conn_register: (0x20002510):   local 0x2000a7b8/log_strdup pool empty!/4242 cb 0x0000894d ud 0x2000005c
~~~

pfalcon, 2018-11-01:
cc: @nordic-krch

jukkar, 2018-11-02:
The log_strdup warning is because CONFIG_LOG_STRDUP_BUF_COUNT is too small. It might not make sense to tweak individual applications because of this, so I suggest that the global default is increased instead. @nordic-krch any opinion about this?

pfalcon, 2018-11-30:
> If <log_strdup alloc failed> message is seen in the log output, it indicates that this value is too small.

That would be the best way to explain it to the end users indeed. The only concern is that the logging message may change over time ;-). But I'd still take what you propose as is, so all that content appeared in Zephyr docs and was searchable by users.

So, I assume we agree and I'll add that to https://github.com/zephyrproject-rtos/zephyr/pull/11735

nordic-krch, 2018-11-30:
@pfalcon 👍

commit 226ff483 (2018-11-30): logging: Elaborate description of CONFIG_LOG_STRDUP_BUF_COUNT

Description is improved, guidelines for when increase is required are
provided, together with associated overheads information.

Fixes: #11007
closed 2018-11-30 by nashif