zephyrproject-rtos/zephyr #11016: nRF52840-PCA10059: Device has empty BD addr when using HCI_USB sample

mike-scott, 2018-11-01:
```
$ lsusb
Bus 002 Device 021: ID 2fe3:0100 

$ hciconfig hci1 commands
hci1:	Type: Primary  Bus: USB
	BD Address: 00:00:00:00:00:00  ACL MTU: 27:7  SCO MTU: 0:0
	Commands: Octet 0  = 0x20 (Bit 5)
		  Octet 2  = 0x80 (Bit 7)
		  Octet 5  = 0xc0 (Bit 6 7)
		  Octet 10 = 0xe4 (Bit 2 5 6 7)
		  Octet 14 = 0x28 (Bit 3 5)
		  Octet 15 = 0x02 (Bit 1)
		  Octet 22 = 0x04 (Bit 2)
		  Octet 25 = 0xf7 (Bit 0 1 2 4 5 6 7)
		  Octet 26 = 0xff (Bit 0 1 2 3 4 5 6 7)
		  Octet 27 = 0xff (Bit 0 1 2 3 4 5 6 7)
		  Octet 28 = 0x0f (Bit 0 1 2 3)
		  Octet 32 = 0x30 (Bit 4 5)
		  Octet 33 = 0xf0 (Bit 4 5 6 7)
		  Octet 34 = 0xf9 (Bit 0 3 4 5 6 7)
		  Octet 35 = 0x7f (Bit 0 1 2 3 4 5 6)
		  Octet 38 = 0x80 (Bit 7)
		  Octet 39 = 0x04 (Bit 2)
	'Disconnect' 'Read Remote Version Information' 'Set Event Mask' 
	'Reset' 'Read Transmit Power Level' 
	'Set Host Controller To Host Flow Control' 'Host Buffer Size' 
	'Host Number Of Completed Packets' 'Read Local Version Information' 
	'Read Local Supported Features' 'Read BD ADDR' 'Set Event Mask Page 2' 
	'LE Set Event Mask' 'LE Read Buffer Size' 
	'LE Read Local Supported Features' 'LE Set Random Address' 
	'LE Set Advertising Parameters' 'LE Read Advertising Channel TX Power' 
	'LE Set Advertising Data' 'LE Set Scan Response Data' 
	'LE Set Advertise Enable' 'LE Set Scan Parameters' 
	'LE Set Scan Enable' 'LE Create Connection' 
	'LE Create Connection Cancel' 'LE Read White List Size' 
	'LE Clear White List' 'LE Add Device To White List' 
	'LE Remove Device From White List' 'LE Connection Update' 
	'LE Set Host Channel Classification' 'LE Read Channel Map' 
	'LE Read Remote Used Features' 'LE Encrypt' 'LE Rand' 
	'LE Start Encryption' 'LE Long Term Key Request Reply' 
	'LE Long Term Key Request Negative Reply' 'LE Read Supported States' 
```

jhedberg, 2018-11-05:
BlueZ knows how to work without a public address. bluetoothd will auto-generate a static random address and keep using that. You do need a fairly recent kernel though. Note that all the legacy commands like hcitool and hciconfig are completely oblivious to LE, which is why e.g. hciconfig only knows about the public address and has no identity address concept (which with these controllers would be a static random address)

jhedberg, 2018-11-05:
Anyway, if you think there's some BlueZ bug I think that'd be best reported to the linux-bluetooth mailing list instead of Zephyr github.

carlescufi, 2018-11-05:
@mike-scott you said that "Using the dongle for IPSP fails". I assume that is not the case with `hci_uart`? So maybe there's something different there, between the initialization done by `btattach` and the one done by the USB class loader? @jhedberg do you know?

mike-scott, 2018-11-05:
@carlescufi yes, running `hci_uart` sample w/ latest Zephyr works, however we do enable additional buffers as such:
```
diff --git a/samples/bluetooth/hci_uart/nrf5.conf b/samples/bluetooth/hci_uart/nrf5.conf
index f2c2ac51f4..02ff4f0c45 100644
--- a/samples/bluetooth/hci_uart/nrf5.conf
+++ b/samples/bluetooth/hci_uart/nrf5.conf
@@ -13,3 +13,10 @@ CONFIG_BT_MAX_CONN=16
 CONFIG_BT_TINYCRYPT_ECC=n
 CONFIG_BT_CTLR_DTM_HCI=y
 CONFIG_BT_CTLR_ASSERT_HANDLER=y
+CONFIG_BT_CTLR_RX_BUFFERS=18
+CONFIG_BT_CTLR_TX_BUFFERS=19
+CONFIG_BT_CTLR_TX_BUFFER_SIZE=251
+CONFIG_BT_CTLR_DATA_LENGTH_MAX=251
+CONFIG_BT_HCI_CMD_COUNT=20
+CONFIG_BT_RX_BUF_COUNT=20
+CONFIG_BT_RX_BUF_LEN=264
```

mike-scott, 2018-11-05:
I updated the title and description of the problem to be more accurate.  This was tested on kernel: 4.18.16

finikorg, 2018-11-09:
Can this be related? Issue #11232

finikorg, 2018-12-03:
Could you check with the fixes for Control endpoint? Basically before you were not able to send more then 64 byte to control endpoint which is HCI commands are using. PR #11713

hasslerb, 2018-12-13:
I experienced the same issue with hciconfig. Instead I needed to bring up the controller with btmgmt per  [Nordic's documentation](https://devzone.nordicsemi.com/b/blog/posts/nrf5x-support-within-the-zephyr-project-rtos).

```
sudo btmgmt --index 0
[hci0]# static-addr FF:02:03:04:05:FF
[hci0]# auto-power
```
After this, hciconfig correctly showed the device as up.

I haven't looked into btmgmt's implementation. What is it doing that hciconfig is not?

jhedberg, 2019-02-01:
I'll assume this as resolved since there's been no update from the reporter. Please re-open if the issue still persists.

closed 2019-02-01 by jhedberg

beikeland, 2020-04-23:
Experiencing the same issue with hci_usb using a nrf52840 dongle and zephyr 2.2.99 `Can't init device hci0: Cannot assign requested address (99)`and setting a static address with `btmgmt` does bring up the device (still showing 00:00:00:00:00:00 in `hciconfig`

hci_usb was just a quick test, aiming to use hci_uart just wanted to report the issue is still present.

kuvaldini, 2020-05-22:
The same issue with latest zephyr [git bc6f11b730 2020-05-21 10:30:18 +0800] and bluez 5.49 on openwrt 18.06

Solved by:
```
# btattach -B /dev/ttyACM0 -S 1000000 -P h4 &
Attaching Primary controller to /dev/ttyACM0
Switched line discipline from 0 to 15
Device index 0 attached
# btmgmt --index 0
[hci0]# static-addr FF:02:03:04:05:FF
Static address successfully set
[hci0]# auto-power
Found controller with index 0
Successfully enabled controller with index 0
[hci0]# find -l
Discovery started
hci0 type 6 discovering on
hci0 dev_found: 63:59:3E:05:43:25 type LE Random rssi -72 flags 0x0000
AD flags 0x1a
eir_len 17
hci0 dev_found: CA:0A:69:8B:37:AB type LE Random rssi -71 flags 0x0000
AD flags 0x06
name DAVINCI IQ
hci0 type 6 discovering off
```

ChristianHirsch, 2020-05-24:
I recently had the same problem on a Raspberry Pi 3, BlueZ 5.50 and a nRF52840 dongle running the HSI_USB sample. When I started the Bluetooth deamon, I saw that it complained about missing crypto implementations. Then, I came across this [article](https://www.bluetooth.com/blog/how-to-setup-bluez-on-raspberry-pi-and-support-pb-adv/), recompiled the Kernel with the specified crypto modules and now everything works fine without setting anything manually.