zephyrproject-rtos/zephyr #11022: mempool allocator is broken

pdunaj, 2018-11-02:
There is a bug in mempool implementation. When memory is allocated allocator gets an available block and if it is large enough it splits these block into 4 smaller blocks. This is repeated until allocator reaches the smallest block that satisfies the request. When `block_free` is called situation is reversed and when 4 neighboring blocks are found to be free allocator concatenates them and returns to the pool as one larger block.

The problem is that to avoid the locking allocator allows a situation when big block or 4 smaller blocks are taken from the pool before the split or merge. Author seemed to know about this limitation and added `-EAGAIN` that would allow alloc to retry but:
- first, when there are no free blocks `-ENOMEM` is returned: even though these blocks may be at the moment be reserved for concatenation `_sys_mem_pool_block_alloc` will not be able to see them, the same happens when `block_free` is merging blocks,
- second, this solution is non-stable: when free is performed by lower priority context and will be interrupted by high priority context just before `block_free` is about to be recursively called, memory will never be reported as free even if we switch to `-EAGAIN` in the `_sys_mem_pool_block_alloc`.

To replicate the problem with higher probability create two context competing for allocation (threads or thread/interrupt) and add busy wait just before `block_free` is recursively called.

There is also question of performance impact. Note that at the moment code is merging smaller blocks in every call to free and break it again with every alloc. Note that in normal situation application will be allocating small blocks only to free them short time later. Wouldn't it be better to merge blocks only when application requests bigger chunk?

I also think that operation of split and merge should be done atomically with free/alloc. If merge is done only on big chunk request the impact may not be big and the code would be stable.

pdunaj, 2018-11-05:
@pizi-nordic

pdunaj, 2018-11-05:
Temporary fix https://github.com/zephyrproject-rtos/zephyr/pull/11106

andrewboie, 2018-11-05:
> There is also question of performance impact. Note that at the moment code is merging smaller blocks in every call to free and break it again with every alloc. Note that in normal situation [...]

> I also think that operation of split and merge should be done atomically with free/alloc. If merge is done only on big chunk request the impact may not be big and the code would be stable.

Can you file enhancements for these two points, and we can use this ticket to track the incorrect behavior you noticed?

andyross, 2018-11-07:
I'll take this one.  @pizi-nordic set me straight on the need, see comments in the linked patch.

andyross, 2018-11-08:
Root cause fix up in #11220, should be good to go though I didn't wait for a full sanitycheck on my end before pushing.

PR #11220 (2018-11-20): lib/mempool: Fix spurious -ENOMEM due to agressive latency control
The mempool operations need to be atomic, but because of latency
concerns (the allocator is intended for use in an ISR) the locking was
designed to be as minimal as possible.  And it... mostly got it right.
All the list handling was correctly synchronized.  The merging of four
child blocks into a parent block was atomic.  The splitting of a block
into four children was atomic.

BUT: there was a moment between the allocation of a large block and
the re-addition of its three unused children where the lock was being
released.  Thi [... cut]
commit 7845e1b0 (2018-11-20): lib/mempool: Fix spurious -ENOMEM due to agressive latency control

The mempool operations need to be atomic, but because of latency
concerns (the allocator is intended for use in an ISR) the locking was
designed to be as minimal as possible.  And it... mostly got it right.
All the list handling was correctly synchronized.  The merging of four
child blocks into a parent block was atomic.  The splitting of a block
into four children was atomic.

BUT: there was a moment between the allocation of a large block and
the re-addition of its three unused children where the lock was being
released.  Th [... cut]
closed 2018-11-20 by carlescufi