zephyrproject-rtos/zephyr #11047: doc: "Power Management Hook Interface" section of "Power Management APIs" is empty

mnkp, 2018-11-02:
"[Power Management Hook Interface](https://docs.zephyrproject.org/latest/api/power_management_api.html#power-management-hook-interface)" section of "[Power Management APIs](https://docs.zephyrproject.org/latest/api/power_management_api.html)" is empty despite the fact that zephyr/include/power.h contains all necessary Doxygen comments.

The likely cause of the issue is power_management_hook_interface Doxygen defgroup being hidden by 
```
#ifdef CONFIG_SYS_POWER_MANAGEMENT
```
which is not enabled by default.

dbkinder, 2018-11-10:
This is a deliberate configuration of doxygen, added by PR #6280

@lpereira @nashif The assumption was that symbols with leading underscores "shouldn't be used by applications, as there's no guarantee they will be available in subsequent versions or maintain
API and ABI compatibility.  Is this still true and the desired behavior of our API doc processing?

nashif, 2018-12-07:
@ramakrishnapallala I do not remember the history behind the naming, but those functions are being documented and appear in a public API, so they should not have _sys prefix. The _ should be removed from all public APIs and documented hooks. underscore should only be used for internal function calls.

PR #11972 (2018-12-28): RFC: Renaming PM APIs
- Rename \_sys to sys\_
- Rename \_sys_soc -> sys\_

Fixes #11047
closed 2018-12-28 by nashif