zephyrproject-rtos/zephyr #11069: Enable instruction and data caches on SAM E70

aurel32, 2018-11-03:
The SAM E70 platform has very bad performances, as the Cortex-M7 data and instruction caches are not enabled. Tentative to enable them have been made in the past, but they caused unexplained crashes in the testsuite.

I am personally not able to reproduce the problem on the current HEAD, but @spoorthik is able to reproduce it. I have done more tests and I have found that the problem is sometimes disappearing for no obvious reason. For example I have the issue with commit f2ba7046eb but not with commit d62e861a01. As we are playing with cache, alignment might play a role if things do not end up in the same cache line. The toolchain probably also plays a role in the resulting alignment, I am using the one from Debian testing.

Anyway working on commits where I have been able to reproduce the issue, I have found the following:

### Way the MPU is disabled and enabled

The `arm_core_mpu_enable()` recently got data and instructions barriers after it (`__DSB()` and `ISB()`). But there is still no barrier for `arm_core_mpu_disable()`. The `Memory Protection Unit for ARM® v8-M based platforms` version 1.1 (I haven't found anything for ARMv7) on the two last pages recommend to use `__DMB()` before disabling the MPU. I have seen improvements in the `mem_protect` tests when using `__DMB()` + `__ISB()`, though some of them still fail.

### Shareability of the RAM regions

The MPU is configured to have the RAM region in WBWA non-shareable (in `include/arch/arm/cortex_m/mpu/arm_mpu_v7m.h`) , but the various memory domains are configured as WBWA shareable (in `arch/arm/core/cortex_m/mpu/arm_mpu_v7_internal.h`). According to the `Table B3.1 ARMv7-M address map` of the `ARM® v7-M ArchitectureReference Manual` (page B3-648), the behaviour when the MPU is disabled is WBWA non-shareable.

For me switching the memory domains as WBWA non-shareable fixes the all issues I am able to reproduce.

### Instruction cache

I have never been able to reproduce any issue with the instruction cache enabled and the data cache disabled. As it already provide a significant memory improvements, I wonder if we can already enable it.

At the end I am using this patch (well in practice a slightly different version that predates the soc/ directory):

```
diff --git a/arch/arm/core/cortex_m/mpu/arm_mpu.c b/arch/arm/core/cortex_m/mpu/arm_mpu.c
index 58ae896612..a12d3e6fd5 100644
--- a/arch/arm/core/cortex_m/mpu/arm_mpu.c
+++ b/arch/arm/core/cortex_m/mpu/arm_mpu.c
@@ -72,6 +72,9 @@ void arm_core_mpu_enable(void)
  */
 void arm_core_mpu_disable(void)
 {
+	__DMB();
+	__ISB();
+
 	/* Disable MPU */
 	MPU->CTRL = 0;
 }
diff --git a/arch/arm/core/cortex_m/mpu/arm_mpu_v7_internal.h b/arch/arm/core/cortex_m/mpu/arm_mpu_v7_internal.h
index 71ff8b5c4e..0264a40c6b 100644
--- a/arch/arm/core/cortex_m/mpu/arm_mpu_v7_internal.h
+++ b/arch/arm/core/cortex_m/mpu/arm_mpu_v7_internal.h
@@ -103,7 +103,7 @@ static inline void _get_mpu_ram_region_attr(arm_mpu_region_attr_t *p_attr,
 	 */
 	(void) base;
 
-	p_attr->rasr = _get_region_attr(1, ap, 1, 1, 1, 1, 0, size);
+	p_attr->rasr = _get_region_attr(1, ap, 1, 1, 1, 0, 0, size);
 }
 
 /**
diff --git a/soc/arm/atmel_sam/same70/soc.c b/soc/arm/atmel_sam/same70/soc.c
index 86ed0d10b3..64391f5831 100644
--- a/soc/arm/atmel_sam/same70/soc.c
+++ b/soc/arm/atmel_sam/same70/soc.c
@@ -227,6 +227,11 @@ static int atmel_same70_init(struct device *arg)
 
 	key = irq_lock();
 
+	SCB_EnableICache();
+	if (!(SCB->CCR & SCB_CCR_DC_Msk)) {
+		SCB_EnableDCache();
+	}
+
 	/* Clear all faults */
 	_ClearFaults(); 
```

@spoorthik, would it please be possible to test that on your board? You might want to test the shareability and the memory barriers separately.

aurel32, 2018-11-04:
I have just tried to define the RAM region as shareable on the SAM E70 board, and the `mem_protect` tests do not even start. Using GDB, it seems they crashes the first time the MPU is enabled. I tried the same thing on an STM32F723 board (after adding `select CPU_HAS_MPU` to Kconfig), and all still works perfectly. This is the patch I used:

```
diff --git a/include/arch/arm/cortex_m/mpu/arm_mpu_v7m.h b/include/arch/arm/cortex_m/mpu/arm_mpu_v7m.h
index 3640a12cb8..696c7ab4e7 100644
--- a/include/arch/arm/cortex_m/mpu/arm_mpu_v7m.h
+++ b/include/arch/arm/cortex_m/mpu/arm_mpu_v7m.h
@@ -103,7 +103,7 @@
 /* Some helper defines for common regions */
 #define REGION_RAM_ATTR(size) \
 { \
-       (NORMAL_OUTER_INNER_WRITE_BACK_WRITE_READ_ALLOCATE_NON_SHAREABLE | \
+       (NORMAL_OUTER_INNER_WRITE_BACK_WRITE_READ_ALLOCATE_SHAREABLE | \
         MPU_RASR_XN_Msk | size | P_RW_U_NA_Msk) \
 }
 #if defined(CONFIG_MPU_ALLOW_FLASH_WRITE)
diff --git a/soc/arm/st_stm32/stm32f7/Kconfig.series b/soc/arm/st_stm32/stm32f7/Kconfig.series
index 38f2648596..69ee295b78 100644
--- a/soc/arm/st_stm32/stm32f7/Kconfig.series
+++ b/soc/arm/st_stm32/stm32f7/Kconfig.series
@@ -9,6 +9,7 @@ config SOC_SERIES_STM32F7X
        bool "STM32F7x Series MCU"
        select CPU_CORTEX_M7
        select CPU_HAS_FPU
+       select CPU_HAS_MPU
        select SOC_FAMILY_STM32
        select SYS_POWER_LOW_POWER_STATE_SUPPORTED
        select HAS_STM32CUBE
```

So it seems using shareable attributes for the RAM is not well supported on the SAM E70 board, while it is correctly supported on the STM32F723 board.

tgorochowik, 2018-11-05:
I think It also might be worth noting that there are GMAC cache problems: #9812. While I did not encounter any problems with caches enabled when I was changing the caches support in SAM GMAC (and manually enabling the actual caches), the remarks by @mnkp seem to be completely valid, so it may be needed to address that too before re-enabling caches on this platform.

However to fix that **properly** - #2927 (or #8784) will be required first.

aurel32, 2018-11-05:
@tgorochowik, I was not aware of this issue thanks for the pointer. It should indeed be fixed, however it should not prevent the instruction cache to be enabled.

As for the data cache, we can also, at least as a first step, enable it only if the GMAC driver is disabled. The other alternative to fix that properly is to use the DTCM memory which doesn't go through the cache.

aurel32, 2018-11-07:
> The MPU is configured to have the RAM region in WBWA non-shareable (in `include/arch/arm/cortex_m/mpu/arm_mpu_v7m.h`) , but the various memory domains are configured as WBWA shareable (`in [...]

Some more details about that part. The Cortex M7 technical reference manual clearly says that cacheable shared locations are treated as being non-cacheable if CACR.SIWT = 0 (default) or as write-through if CACR.SIWT = 1.

First of all it's not what we want. Secondly I guess that when switching from WBWA to nocache (if CACR.SWIT=0) or to WT (if CACR.SIWT = 1), by enabling the MPU after reconfiguring a region or by moving a region from one thread to another, it is needed to at least clean the cache (and possibliy invalidate it when switching to nocache). That might explain the test failures, and why the barriers helps to reduce the number of failures.

aurel32, 2018-11-07:
I have been able to reproduce the issue seen by @spoorthik by using the Zephyr SDK instead of using the toolchain from Debian testing. And that allows me to confirm that the issue is fixed by the above patch. I'll send a proper pull request for it.

aurel32, 2018-11-07:
@tgorochowik, I have just made a proposal for a uncached memory section in #11199

PR #11317 (2018-11-22): Enable instruction and data caches on SAM E70
This patchset enables the instruction and data cache on the Atmel SAM E70 SoC. This has already been tried in the past, but caused issues, and had to be reverted. The issues have been fixed in PR #11194. 

The SAM GMAC Ethernet driver requires non-cached memory. Support for it is being discussed in PR#11199. In the meantime, do not enable the data cache if the SAM GMAC Ethernet driver is enabled.

Fixes #11069.
closed 2018-11-22 by nashif