zephyrproject-rtos/zephyr #11074: samplesapplication_development

zhang-wenchao, 2018-11-04:
cd samples/application_development/external_lib

mkdir build && cd build

cmake -GNinja -DBOARD=frdm_k64f ..
```
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.7", minimum required is "3.4")
-- Selected BOARD frdm_k64f
Zephyr version: 1.13.99
Parsing Kconfig tree in /Users/wenchao/Application/Zephyr/Kconfig
Loading /Users/wenchao/Application/Zephyr/boards/arm/frdm_k64f/frdm_k64f_defconfig as base
Merging /Users/wenchao/Application/Zephyr/samples/application_development/external_lib/prj.conf

-- Generating zephyr/include/generated/generated_dts_board.h
/Users/wenchao/Application/Zephyr/scripts/dts/extract_dts_includes.py:482: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
  and isinstance(merge_dct[k], collections.Mapping)):
extract_dts_includes.py: Merge of 'category': 'required'  overwrites 'optional'.
extract_dts_includes.py: Merge of 'category': 'required'  overwrites 'optional'.
-- Cache files will be written to: /Users/wenchao/Library/Caches/zephyr
-- The C compiler identification is GNU 7.3.1
-- The CXX compiler identification is GNU 7.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /Users/wenchao/Application/gcc-arm-none-eabi/bin/arm-none-eabi-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/wenchao/Application/Zephyr/samples/application_development/external_lib/build
```
ninja

```
[1/115] Preparing syscall dependency handling

[15/115] Performing build step for 'mylib_project'
mkdir -p /Users/wenchao/Application/Zephyr/samples/application_development/external_lib/build/mylib/obj /Users/wenchao/Application/Zephyr/samples/application_development/external_lib/build/mylib/lib
/Users/wenchao/Application/gcc-arm-none-eabi/bin/arm-none-eabi-gcc -c -I/Users/wenchao/Application/Zephyr/kernel/include -I/Users/wenchao/Application/Zephyr/arch/arm/include -I/Users/wenchao/Application/Zephyr/soc/arm/nxp_kinetis/k6x -I/Users/wenchao/Application/Zephyr/soc/arm/nxp_kinetis/k6x/include -I/Users/wenchao/Application/Zephyr/soc/arm/nxp_kinetis/include -I/Users/wenchao/Application/Zephyr/boards/arm/frdm_k64f -I/Users/wenchao/Application/Zephyr/include -I/Users/wenchao/Application/Zephyr/include/drivers -I/Users/wenchao/Application/Zephyr/samples/application_development/external_lib/build/zephyr/include/generated -I/Users/wenchao/Application/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/7.3.1/include -I/Users/wenchao/Application/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/7.3.1/include-fixed -I/Users/wenchao/Application/Zephyr/lib/libc/minimal/include -I/Users/wenchao/Application/Zephyr/ext/hal/cmsis/Include -I/Users/wenchao/Application/Zephyr/ext/hal/nxp/mcux/devices/MK64F12 -I/Users/wenchao/Application/Zephyr/ext/hal/nxp/mcux/drivers/kinetis/. -DKERNEL -D__ZEPHYR__=1 -D_FORTIFY_SOURCE=2 -DBUILD_VERSION=zephyr-v1.13.0-1548-ge4aa946c55 -DCPU_MK64FN1M0VLL12 -Os -g -Wall -Wformat -Wformat-security -Wno-format-zero-length -imacros /Users/wenchao/Application/Zephyr/samples/application_development/external_lib/build/zephyr/include/generated/autoconf.h -ffreestanding -Wno-main -mthumb -mcpu=cortex-m4 -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -Wno-pointer-sign -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Werror=implicit-int -Wpointer-arith -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -isystem/Users/wenchao/Application/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/7.3.1/include -isystem/Users/wenchao/Application/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/7.3.1/include-fixed -Iinclude src/mylib.c -o /Users/wenchao/Application/Zephyr/samples/application_development/external_lib/build/mylib/obj/mylib.o
ar -rcs /Users/wenchao/Application/Zephyr/samples/application_development/external_lib/build/mylib/lib/libmylib.a /Users/wenchao/Application/Zephyr/samples/application_development/external_lib/build/mylib/obj/mylib.o
warning: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: warning for library: /Users/wenchao/Application/Zephyr/samples/application_development/external_lib/build/mylib/lib/libmylib.a the table of contents is empty (no object file members in the library define global symbols)
[110/115] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf
: && ccache /Users/wenchao/Application/gcc-arm-none-eabi/bin/arm-none-eabi-gcc   -L/usr/local/lib -L/usr/local/opt/openssl/lib -L/usr/local/opt/openssl@1.1/lib zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -T zephyr/linker.cmd -Wl,-Map=/Users/wenchao/Application/Zephyr/samples/application_development/external_lib/build/zephyr/zephyr.map -u_OffsetAbsSyms -u_ConfigAbsSyms -Wl,--whole-archive libapp.a zephyr/libzephyr.a zephyr/arch/arm/core/libarch__arm__core.a zephyr/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a zephyr/arch/arm/core/cortex_m/mpu/libarch__arm__core__cortex_m__mpu.a zephyr/lib/libc/minimal/liblib__libc__minimal.a zephyr/boards/arm/frdm_k64f/libboards__arm__frdm_k64f.a zephyr/drivers/gpio/libdrivers__gpio.a zephyr/drivers/serial/libdrivers__serial.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -L"/Users/wenchao/Application/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/7.3.1/thumb/v7e-m" -L/Users/wenchao/Application/Zephyr/samples/application_development/external_lib/build/zephyr -lgcc -Wl,--print-memory-usage mylib/lib/libmylib.a -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--gc-sections -Wl,--build-id=none -Wl,--orphan-handling=warn && :
Memory region         Used Size  Region Size  %age Used
           FLASH:       14344 B         1 MB      1.37%
            SRAM:        4076 B       192 KB      2.07%
        IDT_LIST:          88 B         2 KB      4.30libapp.a(main.c.obj): In function `main':
/Users/wenchao/Application/Zephyr/samples/application_development/external_lib/build/../src/main.c:18: undefined reference to `mylib_hello_world'
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.
```

zhang-wenchao, 2018-11-04:
macOS 10.14

zhang-wenchao, 2018-11-04:
I tried to use a static library, always wrong, until I tried the samples , I got an error.
@SebastianBoe What is the cause of this, looking forward to answering.

SebastianBoe, 2018-11-05:
This is how 'ar' is discovered:

find_program(CMAKE_AR         ${CROSS_COMPILE}ar      PATH ${TOOLCHAIN_HOME} NO_DEFAULT_PATH)

It is supposed to find the cross-compiling toolchain's 'ar'.

Are you able to find out why it is not doing so?

If we knew the root-cause I might be able to create a fix.

zhang-wenchao, 2018-11-05:
I am trying to fix this problem

zhang-wenchao, 2018-11-06:
I have found the problem, fixing it, please wait for my pr

PR #11161 (2018-11-07): samples:application_development: fix #11074 
fix #11074

using the wrong ar

Missing ar configuration
commit 93d30652 (2018-11-07): samples: application_development: fix #11074

external_lib using the wrong ar
closed 2018-11-07 by carlescufi