zephyrproject-rtos/zephyr #11077: Compiler warning at include/misc/util.h when building for ESP32

nniranjhana, 2018-11-05:
We have a compiler warning being treated as error, thus stopping build.

The issue as reported says that 

` #define BIT(n)  (1UL << (n))`

is re-defined at _/zephyr/include/misc/util.h:135_

Well, it complains because when we build with our toolchains, **esp-idf** already includes it at 
_esp-idf/components/soc/esp32/include/soc/soc.h:128_

as `#define BIT(nr)                 (1UL << (nr))`

So we should probably not include our definition when building for ESP32.

pfalcon, 2018-11-06:
@lpereira : Is there a reason why you're not in CODEOWNERS for any xtensa/esp32 stuff?

lpereira, 2018-11-06:
@pfalcon I'm not working on Zephyr any longer.

lpereira, 2018-11-06:
In any case, this is just the matter of including headers in the right order.  This error happened before; see, for instance, `drivers/serial/uart_esp32.c`.

nashif, 2019-01-08:
not fixed yet

nashif, 2019-01-11:
@dcpleung you introduced this issue in 5a47c60dbf4f24fa9ad2206502d5eab152587eac, can you please take a look.

PR #12445 (2019-01-11): esp32: fix build error regarding multiplt BIT() definitions
This partially reverts commit 5a47c60dbf4f24fa9ad2206502d5eab152587eac.
The soc.h is now only included when _soc_irq_*() is being referred.

Fixes #11077.
commit c17d0dc2 (2019-01-11): esp32: fix build error regarding multiplt BIT() definitions

This partially reverts commit 5a47c60dbf4f24fa9ad2206502d5eab152587eac.
The soc.h is now only included when _soc_irq_*() is being referred.

Fixes #11077.
closed 2019-01-11 by andrewboie