zephyrproject-rtos/zephyr #11078: Kconfig/Cmake not aligned

nordic-krch, 2018-11-05:
Sample: net/http_server for 96b_carbon board with overlay-netusb.conf.

Following warning during CMake:

```
warning: NET_6LO (defined at subsys/net/ip/Kconfig.ipv6:121) has direct dependencies NET_IPV6 && !NET_RAW_MODE && NETWORKING with value n, but is currently being y-selected by the following symbols:
 - NET_L2_BT (defined at /home/user/zephyr/boards/arm/96b_carbon/Kconfig.defconfig:70, subsys/net/l2/Kconfig:16), with value y, direct dependencies (NETWORKING && BOARD_96B_CARBON) || NETWORKING (value: y), and select condition NETWORKING (value: y)
```

As the outcome NET_6LO is set and 6lo.c file is compiled (CMakeLists.txt: `zephyr_library_sources_ifdef(CONFIG_NET_6LO          6lo.c)`) but Kconfig options are cut because in kconfig NET_6LO options depends on NET_IPV6 option which is disabled.

I'm not sure if this is framework error or board configuration error.
Note that this error is not visible on master but appeared on branch which is using one of the kconfig flag which is not present (https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/25994/1/tests).

nordic-krch, 2018-11-05:
@jukkar could you advise? It may be that this is wrong board configuration?
It seems like 96b_carbon board may have error:

```
# BT is the only onboard network iface, so use it for IP networking
# if it's enabled

config NET_L2_BT
	def_bool y
```

jukkar, 2018-11-05:
> because in kconfig NET_6LO options depends on NET_IPV6 option which is disabled.

This is highly suspicious because IPv6 is needed by 6lo and by Bluetooth. Something is not selecting or depending on CONFIG_NET_IPV6

nordic-krch, 2018-11-05:
It's because sample is deselecting IPV6:
https://github.com/zephyrproject-rtos/zephyr/blob/6203020bfd7985eae9c7f6de02247d937fdd28af/samples/net/http_server/overlay-netusb.conf#L26

jukkar, 2018-11-05:
In that case NET_L2_BT should not be selected as the IPv6 support will not be there and BT IPSP will not work, which we saw here because of compilation error. The board is still usable, it is just that the Bluetooth IP connectivity will not work.

nordic-krch, 2018-11-05:
do you suggest to add `NET_L2_BT=n` to netusb overlay?

jukkar, 2018-11-05:
I was thinking more like adding `depends on NET_IPV6` to where NET_L2_BT is set. For example to subsys/net/l2/Kconfig where NET_L2_BT is defined.

nordic-krch, 2018-11-05:
ok, I'll fix that as the part of #11035 
https://github.com/zephyrproject-rtos/zephyr/pull/11035/commits/bb6c63a4e1532133e123d42630d440c236de4af3

PR #11035 (2018-11-10): logging: Rafactor LOG_MODULE_REGISTER and LOG_MODULE_DECLARE macros
Changed LOG_MODULE_REGISTER and LOG_MODULE_DECLARE macros to take log
level as optional parameter. LOG_MODULE_DECLARE can now also be used
in static inline functions in headers. Added LOG_LEVEL_SET macro
which is used when instance logging API is used to indicate maximal
log level compiled into the file.

This is based on #10215 but LOG_MODULE_REGISTER/LOG_MODULE_DECLARE have been made backward compatible to be able to use following:

```
#define LOG_LEVEL CONFIG_LEVEL
LOG_MODULE_REGISTER(name);
```

or
```
LOG_MODULE_REGISTER [... cut]
closed 2018-11-10 by nashif